pylessard/python-udsoncan #110: Support DTC User Defined Memory

kayoub5, 2022-02-21:
From https://piembsystech.com/read-dtc-information-service-0x19-uds-protocol/

0x17 | Report User Defined Memory DTC By Status Mask |  
-- | -- | --
0x18 | Report User Defined Memory DTC Snapshot Record By DTC Number |  
0x19 | Report User Defined Memory DTC Extended Data Record By DTC Number

pylessard, 2022-02-21:
Oh, these are news. 
I will check this. Hopefully, it should be easy to integrate.

Regards

pylessard, 2022-03-19:
Hi @kayoub5 ,
Can you have a look at this branch?  [110-support-dtc-user-defined-memory](https://github.com/pylessard/python-udsoncan/tree/110-support-dtc-user-defined-memory)

I've added 3 methods to the client object.

``get_user_defined_memory_dtc_by_status_mask``
``get_user_defined_dtc_snapshot_by_dtc_number``
``get_user_defined_dtc_extended_data_by_dtc_number``

Let me know if they answer you needs.  I do have other subfunction to support as 2020 version added lot of them!

kayoub5, 2022-03-24:
Hi @pylessard, the ECU accepted the requests made by the new functions, but the response always had no data, not sure if it was something to do with the decoding logic, or the ECU actually have no data.

Currently checking.

pylessard, 2022-03-25:
Hi, you can share the log. If no exception is raised, the response should be valid. So probably there really is no data

yassin92, 2022-05-04:
Hi @pylessard get_user_defined_memory_dtc_by_status_mask and get_user_defined_dtc_snapshot_by_dtc_number was tested on real HW ,where we have DTCs in user defined memory , and provided the expected results.

pylessard, 2022-05-23:
Yes definetely, I was hoping to add other subfunction as well in the next release but i'm putting priority on a new project. If you need it quick, I can release now and push the other subfunctions to the one after this.

Regards

kayoub5, 2022-05-23:
> Yes definetely, I was hoping to add other subfunction as well in the next release but i'm putting priority on a new project. If you need it quick, I can release now and push the other subfunctions to the one after this.
> 
> Regards

Releasing now would be nice, other functions can come later.

pylessard, 2022-05-23:
Will do soon.

closed 2022-05-29 by pylessard

pylessard, 2022-05-29:
Here you go sirs
https://github.com/pylessard/python-udsoncan/releases/tag/v1.15

kayoub5, 2022-05-29:
@pylessard thanks