pylessard/python-udsoncan #112: Return None but the log file shown the CAN channel response

SiamakMirifar, 2022-03-13:
I am using udsoncan python library to read/write to CANtp (vector CANoe). I can send data to the board, however, when I read data it returns None value. When I check the log file I see that the return value exists in the log but not when I want to print it.

I thought maybe because I am logging the output It might return None, As on logging it read the data. I have to commend the log. But even after commenting on the log, I get None as a return.

Note: The Board just returns a signal on a Tx channel once it receives data on the Rx channel, after that it returns once it does not return anything!!

Here is my code:
`class TestCanTp:
     def __init__(self):
         self.bus = can.interface.bus('vector', '0', 'CANoe')
         self.addr = isotp.address(addressing_mode = isotp.AddressingMode.Normal_11bits, txid = 0x106, rxid = 0x186)
         self.stack = isotpCanStack(self.bus, self.addr, isotp_params)
         self.udscan = PythonIsoTpConnection(self.stack)
     
     def diagnostic_req(self, payload):
         self.udscan.send(payload)
    
     def diagnostic_res(self)
         return self.udscan.wait_frame(timeout = 2)

test = TestCanTp()
test.diagnostic_req(payload)
print(test.diagnostic_res())`

pylessard, 2022-03-15:
Hi,
The Connection object is a bridge between the transport layer and the application layer. In your case, you seem to send raw data without making usage of the Client object. Normally, this connection needs to be "opened" before used, so your code should throw an exception since you didn't call "open()" before calloing `wait_frame()`, if it doesn't maybe you run on a old version of the lib. latest is at v1.14

In any case, if you want to send raw data, you do not need `udsoncan`, you can interact directly with the CanStack object. Otherwise, you can uses the `udsoncan.Client` 

If you want more help, you can share a bigger snippet of code and also a CAN log.

SiamakMirifar, 2022-03-15:
Hi, thank you for your response. 

What i am gonna to do is to update the board with using CANtp protocol and diagnostic channel. I have to send a payload almost 250 bytes in each message. Currently the board is under develop and does not accept more than 8 bytes through can so i config my current code to send 8 byte payload.
When i use stack, the board does not receive message! But through udsoncan i can send the payload and shows in log that the board send response ! But my return stays as None.

I attached the images of my code and log!

[image]

[image]

[image]

SiamakMirifar, 2022-03-15:
> Hi,
>
> The Connection object is a bridge between the transport layer and the application layer. In your case, you seem to send raw data without making usage of the Client object. Normally, this [...]

Hi, i have check the source code and i assume that after calling self.stack.process() the logger actually get the msg and therefore the recv() does not return, i am correct ? If yes, so how can i implement the send and recv()

pylessard, 2022-03-16:
Hi,
The recv function is non-blocking and therefore returns None if no answer has been received yet.  
Wait for `available()` to return True before calling `recv`.
See the documentation.
https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#usage

You don't have to bother about `process_rx` and `process_tx`.  You must call `process` as fast as you can. Everything else is handled. You wait for `available()` to be True then read the message queue.

Good luck

SiamakMirifar, 2022-03-16:
> Hi, The recv function is non-blocking and therefore returns None if no answer has been received yet. Wait for `available()` to return True before calling `recv`. See the documentation. [...]

That is my problem, when I call process it read the queue and then when I call the recv() it return None. The board sending other messages also from other CAN channel. So when I call process it send and read everything ( look at the log please ), therefore I think recv() is None.

Using available() is also did not help because it is always False after if call process. Is there any way just to process the send command. And recv() them from another function?

pylessard, 2022-03-16:
The TransportLayer will drop any messages that doesn't match the address. If you don't receive any frame, something else is wrong.

You can share a complete CAN log and code maybe. Not pictures of your screen with partial info.

SiamakMirifar, 2022-03-16:
> The TransportLayer will drop any messages that doesn't match the address. If you don't receive any frame, something else is wrong.
> 
> You can share a complete CAN log and code maybe. Not pictures of your screen with partial info.

Here you can see my code. The commented code are my tries to find other possibilities
```
class testCanTP():
    def __init__(self) -> None:
        try:
            self.bus = can.interface.Bus(bustype='vector', channel='0', app_name='CANoe', bitrate=1000000)
            self.addr = isotp.Address(addressing_mode=isotp.AddressingMode.Normal_11bits, txid= 0x601 , rxid= 0x681)  # Base: txid=0x601, rxid=0x681
            self.transportLayer = isotp.TransportLayer(rxfn = my_rxfn,txfn =  my_txfn, address= self.addr, params= isotp_params) # , params= isotp_params
            self.stack = isotp.CanStack(bus=self.bus, address=self.addr, params=isotp_params) #  , TransportLayer= self.transportLayer
            self.conn = PythonIsoTpConnection(self.stack)
            
        except:
            AssertionError(sys.exc_info()[0])
            logger.info("Can config FAILED!")
        else:
            logger.info("Can is Configured")
    
    def diagnostic_request(self, request_dat):
        request_pdu = binascii.a2b_hex(request_dat)

        try:
            # self.transportLayer.send(request_dat)
            # self.conn.specific_send(request_pdu) # changed to no input 
            # self.conn.send(request_pdu)
            self.stack.send(request_pdu)
            # self.transportLayer.send(request_pdu)
            

            bCon1 = True
            if bCon1:
                while self.stack.transmitting():

                    self.stack.process()
                    time.sleep(self.stack.sleep_time())

            bCon3 = False
            if bCon3:
                while self.transportLayer.transmitting():
                    self.transportLayer.process()
                    time.sleep(self.transportLayer.sleep_time())

        except can.CanError:
            AssertionError("Send request data fail") 
    
    def diagnostic_respond(self):

        if not self.conn.is_open():
            self.conn.open()

            try:
                # resp_pdu = self.conn.wait_frame(timeout=2)
                # resp_pdu = self.conn.specific_wait_frame(timeout=2)
                resp_pdu = self.stack.recv()
                # resp_pdu = self.transportLayer.recv()
            except Exception as err:
                raise err

            if resp_pdu is not None:
                #logger.info('ECU: Received %d bytes : <%s>' % (len(resp_pdu), binascii.hexlify(resp_pdu)))
                pass

        return resp_pdu

    def close_can(self):
        self.conn.close()
        self.bus.stop_all_periodic_tasks()
[... 77 lines cut ...]
13:34:21,116 isotp DEBUG Sending : <601> (8)	 b'2330303004330380'
13:34:21,132 isotp DEBUG Sending : <601> (8)	 b'24d0a3032100003a'
13:34:21,179 isotp DEBUG Receiving : <201> (6)	 b'11520300808b'
13:34:21,179 isotp DEBUG Receiving : <211> (8)	 b'1000000000000000'
13:34:21,179 isotp DEBUG Receiving : <681> (8)	 b'300a0c0000000000'
13:34:21,179 isotp DEBUG Receiving : <701> (7)	 b'73312121787302'
13:34:21,180 isotp DEBUG Receiving : <711> (4)	 b'07060000'
13:34:21,196 isotp DEBUG Sending : <601> (8)	 b'2532332303030303'
13:34:21,211 isotp DEBUG Receiving : <702> (4)	 b'14215a00'
13:34:21,211 isotp DEBUG Receiving : <212> (8)	 b'0800000000000000'
13:34:21,211 isotp DEBUG Receiving : <202> (4)	 b'11001408'
13:34:21,213 isotp DEBUG Sending : <601> (8)	 b'2600303000200002'
13:34:21,228 isotp DEBUG Sending : <601> (8)	 b'27000a4346353032'
13:34:21,243 isotp DEBUG Sending : <601> (8)	 b'2830303004330380'
13:34:21,259 isotp DEBUG Sending : <601> (8)	 b'29d0a3032100003a'
13:34:21,274 isotp DEBUG Receiving : <201> (6)	 b'11520300808b'
13:34:21,275 isotp DEBUG Receiving : <211> (8)	 b'1000000000000000'
13:34:21,275 isotp DEBUG Receiving : <701> (7)	 b'73312121787302'
13:34:21,275 isotp DEBUG Receiving : <711> (4)	 b'07060000'
13:34:21,277 isotp DEBUG Sending : <601> (3)	 b'2a3230'
13:34:21,338 isotp DEBUG Receiving : <702> (4)	 b'1521be00'
13:34:21,338 Connection INFO Connection opened
13:34:21,338 isotp DEBUG Receiving : <202> (4)	 b'11001408'
13:34:21,399 Connection INFO Connection closed
```

pylessard, 2022-03-16:
You don't wait for `available()`

Do something like : 

    while True:
        stack.process()
        if stack.available():
            frame = stack.recv()
            print('Got frame : %s' % frame)
        time.sleep(stack.sleep_time())

pylessard, 2022-03-16:
oh, and you receive nothing on 681. So that could explain you problems ;)

pylessard, 2022-03-16:
Good news then. Shall we close this issue?

closed 2022-03-16 by pylessard