pylessard/python-udsoncan #113: make_request code problems

feuerball, 2022-03-15:
I'm not sure if im understanding something wrong in here, but in my opinion the try except is completely obsolete nor?
File is services/ReadDataByIdentifier
```

        did_reading_all_data = None
        for did in didlist:
            if did not in didconfig:    # Already checked in check_did_config. Paranoid check
                raise ConfigError(key=did, msg='Actual data identifier configuration contains no definition for data identifier 0x%04x' % did)
                
            codec = DidCodec.from_config(didconfig[did])
            try:
                length = len(codec)
                if did_reading_all_data is not None:
                    raise ValueError('Did 0x%04X is configured to read the rest of the payload (__len__ raisong ReadAllRemainingData), but a subsequent DID is requested (0x%04x)' % (did_reading_all_data, did))
            except DidCodec.ReadAllRemainingData:
                if did_reading_all_data is not None:
                    raise ValueError('It is impossible to read 2 DIDs configured to read the rest of the payload (__len__ raising ReadAllRemainingData). Dids are : 0x%04X and 0x%04X' % (did_reading_all_data, did))
                did_reading_all_data = did

```

pylessard, 2022-03-15:
Hi,
No, it is not useless. It's to handle the case where a codec wants to read all the remaining data which is achieved by raising ReadAllRemainingData exception in the __len__ function. 

This can only be achieved for the last did of a list, hence the validation you see.

This feature is not defined by Iso14229, but has been requested by a user

pylessard, 2022-03-15:
https://udsoncan.readthedocs.io/en/latest/udsoncan/helper_classes.html#didcodec

feuerball, 2022-03-16:
Ah okay, thanks for the explanation. 

Wasn't expecting that someone trys to use exception from the len function :)

closed 2022-03-16 by feuerball