pylessard/python-udsoncan #114: UdsClient: [AttributeError] : 'tuple' object has no attribute 'encode'

catmemo, 2022-03-24:
Hi,

I tried to write value by client.write_data_by_identifier(did, value) with did = 0xF184 and value = ('FF', '22', '03', '24', 'FF', 'FF', 'FF', 'FF', 'FF', 'FF') but got failed.

File "D:\Development_Tools\Python\lib\site-packages\udsoncan\client.py", line 131, in decorated
    return func(self, *args, **kwargs)
  File "D:\Development_Tools\Python\lib\site-packages\udsoncan\client.py", line 436, in write_data_by_identifier
    req = services.WriteDataByIdentifier.make_request(did, value, didconfig=self.config['data_identifiers'])
  File "D:\Development_Tools\Python\lib\site-packages\udsoncan\services\WriteDataByIdentifier.py", line 44, in make_request
    req.data += codec.encode(value)
  File "D:\Development_Tools\Python\lib\site-packages\udsoncan\__init__.py", line 97, in encode
    return string_ascii.encode('ascii')
AttributeError: 'tuple' object has no attribute 'encode'

pylessard, 2022-03-24:
Why don't you share your configuration. Looking at the stack trace, the error is raised in the codec you wrote. Apparently, you try to read ascii but receive non-ascii chars

catmemo, 2022-03-24:
Hello, 
As you see below is my configuration:
`isotp_params = {
    'stmin': 32,  # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
    'blocksize': 8,  # Request the sender to send 8 consecutives frames before sending a new flow control message
    'wftmax': 0,  # Number of wait frame allowed before triggering an error
    'll_data_length': 8,  # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
    'tx_padding': 0,  # Will pad all transmitted CAN messages with byte 0x00. None means no padding
    'rx_flowcontrol_timeout': 1000,  # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
    'rx_consecutive_frame_timeout': 1000,  # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
    'squash_stmin_requirement': False  # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
}
global config
config = dict(udsoncan.configs.default_client_config)
config['data_identifiers'] = {
    0xF184: udsoncan.AsciiCodec(10),
}

tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x72b, rxid=0x7ab)  # Network layer addressing scheme
stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)  # Network/Transport layer (IsoTP protocol)
conn = PythonIsoTpConnection(stack)
`

pylessard, 2022-03-24:
Exactly, you use the AsciiCodec for non-ascii characters. That is tour issue.  Ascii chars range from 0 to 0x7F

catmemo, 2022-03-24:
> AsciiCodec
If I changed it to `0xF184: udsoncan.DidCodec(10)` then got another error:

 File "D:\Live_And_Learn\Python\flash\flash_gac\Send_Diag_Command.py", line 212, in send_diag_write_command
    write_response = client.write_data_by_identifier(int(did, 16), value)
  File "D:\Development_Tools\Python\lib\site-packages\udsoncan\client.py", line 131, in decorated
    return func(self, *args, **kwargs)
  File "D:\Development_Tools\Python\lib\site-packages\udsoncan\client.py", line 436, in write_data_by_identifier
    req = services.WriteDataByIdentifier.make_request(did, value, didconfig=self.config['data_identifiers'])
  File "D:\Development_Tools\Python\lib\site-packages\udsoncan\services\WriteDataByIdentifier.py", line 44, in make_request
    req.data += codec.encode(value)
  File "D:\Development_Tools\Python\lib\site-packages\udsoncan\__init__.py", line 53, in encode
    return struct.pack(self.packstr, *did_value)
TypeError: Struct() argument 1 must be a str or bytes object, not int

catmemo, 2022-03-25:
> What do you expect as a result? You codec definition still makes no sense. Do you want raw data or decoded value somehow?

Sorry but I'm not so familiar with CAN.
What I want to do is to write values with 0xF184 and it seems that I have to add the DID to config['data_identifiers'].
But the data type of 0xF184 are not ASCII:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///D:/Users/F47920C/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///D:/Users/F47920C/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<style>

</style>
</head>

<body link=blue vlink=purple>

0xF184 | Unsigned
-- | --
BCD
BCD
BCD
Unsigned
Unsigned
Unsigned

</body>

</html>

closed 2022-03-30 by catmemo

lanwailan, 2024-07-29:
how did you fix this?

lanwailan, 2024-07-29:
> Solved by customizing a class finally. So close this issue~

I met same problems, how did you fix this problem?

pylessard, 2024-07-29:
If you encounter these type of issue, the Did configuration is certainly wrong. If you share, we can help.
Otherwise, check the doc to know how to configure it