pylessard/python-udsoncan #116: Functionality for CAN-FD

Ahelion, 2022-03-28:
General question(not quite issues):
Background: I can set up the Bus ( vector ) as CAN_FD, but when the message is send via client.read_data_by_id the Baud Rate Switch BIT is not set.
Is there something extra to do for the client?
Checked with CANOE for logging only.

pylessard, 2022-03-28:
Do you use Python-can?  If you do, you must set the flag is_fd=true when sending a message.  If you use the PythonCanIsoTPconenction, there's an option in the TransportLayer to automatically set that flag. 

See : https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#can_fd

pylessard, 2022-03-28:
You must set the option on the TransportLayer. I gave you a link to the aprameter definition. Read what I wrote.

pylessard, 2022-03-28:
With python-can, you can use a Can-fd bus to send normal message. So you also need to set a flag on each message. I think this is what confuses you.

Ahelion, 2022-04-01:
I did not forget, just waiting for HW to free up, for me to test it.

pylessard, 2022-05-29:
shall we close this issue?

closed 2022-07-07 by pylessard

Ahelion, 2022-07-14:
> shall we close this issue?

YES! thank you very much