pylessard/python-udsoncan #124: Listener/Notifier and UDS command

jennapark86, 2022-06-24:
Hello, when i have a logger and a notifier on the can layer, it seems like uds command (e.g. ecu reset) times out and never returns response. My code is below. Is there a way to have a logger constantly logging CAN messages while sparingly sending UDS commands? Uncomment initialization of logger1 and notifier1 on line 17 and 18. 

my setup is Peak's pcan usb dongle with an ecu with uds server. 

[TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=2.000 sec)

import can
from can.interfaces.pcan.pcan import PcanBus
import udsoncan
import isotp
from udsoncan.connections import IsoTPSocketConnection, PythonIsoTpConnection
from udsoncan.client import Client
from udsoncan.exceptions import *
from udsoncan.services import *

import time
import os

if __name__ == "__main__":

    can_bus = PcanBus("PCAN_USBBUS7", bitrate=500000)
    #logger1 = can.Logger("test.asc")
    #notifier1 = can.Notifier(bus=can_bus, listeners=[logger1])

     
    
    isotp_params = {
   'stmin' : 32,                          # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
   'blocksize' : 8,                       # Request the sender to send 8 consecutives frames before sending a new flow control message
   'wftmax' : 0,                          # Number of wait frame allowed before triggering an error
   'tx_data_length' : 8,                  # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
   'tx_data_min_length' : None,           # Minimum length of CAN messages. When different from None, messages are padded to meet this length. Works with CAN 2.0 and CAN FD.
   'tx_padding' : 0,                      # Will pad all transmitted CAN messages with byte 0x00.
   'rx_flowcontrol_timeout' : 1000,       # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
   'rx_consecutive_frame_timeout' : 1000, # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
   'squash_stmin_requirement' : False,    # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
   'max_frame_size' : 4095                # Limit the size of receive frame.
    }
    config={
    "standard_version": 2006,
    "use_server_timing": False,
    "ignore_all_zero_dtc": True,
    "tolerate_zero_padding": True,
    "p2_timeout": 10,
    }
    tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x604, rxid=0x614) # Network layer addressing scheme
    stack = isotp.CanStack(bus=can_bus, address=tp_addr, params=isotp_params)               # Network/Transport layer (IsoTP protocol)
    conn = PythonIsoTpConnection(stack)  
    client = Client(conn, request_timeout=2, config=config)

    with Client(conn, request_timeout=2, config=config) as client:                                     # Application layer (UDS protocol)
        client.ecu_reset(ECUReset.ResetType.hardReset)

    #notifier1.stop()
    #logger1.stop()

    can_bus.shutdown()

pylessard, 2022-06-24:
It's a common problem. There are several similar issues. I do not know the can.notifier object, but it must be consuming the data, leaving nothing for the TransportLayer

If the can messages needs to be sent to 2 independent objects, you have to have a software layer that consume the data and make as much copy that is needed for each objects.  You can write your own  `txfn` and `rxfn` functions that will fetch the can messages from your dispatching layer.

I will eventually write an example for this in the doc

pylessard, 2022-06-25:
So I read about notifiers and they serm yo do exactly that. The CanStack object should register a Listener instead of reading directly the bus object and consuming the buffer 

I will move this issue to the can-isotp project and do the improvement when I get time

jennapark86, 2022-06-27:
thanks for looking into this! i'd love to be able to continuously log CAN data while being able to send diagnostic commands