pylessard/python-udsoncan #126: timeout,  Did not received ISOTP frame in time

ziyangfu, 2022-07-08:
we use IsoTPSocketConnection
why timeout value can not reset to 2 when continue to receive the frame ?
when I use service(0x34), the ECU take a lot of time to prepare and continue to response 0x78 before give a positive
response.
in this period, udsoncan exit because of timeout
```
# connections.py  Line 314
 def specific_wait_frame(self, timeout=2):
        if not self.opened:
            raise RuntimeError("Connection is not open")

        timedout = False
        frame = None
       
        print("before timeout = %s " % timeout)    # for test
        try:
            frame = self.rxqueue.get(block=True, timeout=timeout)

        except queue.Empty:
            timedout = True
            print("after exception timeout = %s " % timeout)     # for test

        if timedout:
            raise TimeoutException("Did not received ISOTP frame in time (timeout=%s sec)" % timeout)

        return frame
```

error log :
```
2022-07-08 16:47:27 [DEBUG] Connection: Sending 11 bytes : [b'340044XXXXXXXXXXXXX']
before timeout = 1.9999797344207764 
2022-07-08 16:47:27 [DEBUG] Connection: Received 3 bytes : [b'7f3478']
before timeout = 1.9897913932800293 
2022-07-08 16:47:28 [DEBUG] Connection: Received 3 bytes : [b'7f3478']
before timeout = 0.45956945419311523 
after exception timeout = 0.45956945419311523 
2022-07-08 16:47:29 [DEBUG] Connection: No data received: [TimeoutException] - Did not received ISOTP frame in time (timeout=0.45956945419311523 sec) 
2022-07-08 16:47:29 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=0.460 sec)
2022-07-08 16:47:29 [INFO] Connection: Connection closed
```
my temporary solution (do not suggestion)
```
def specific_wait_frame(self, timeout=2):
        if not self.opened:
            raise RuntimeError("Connection is not open")

        timedout = False
        frame = None
        timeout = 2   # look here
        try:
            frame = self.rxqueue.get(block=True, timeout=timeout)

        except queue.Empty:
            timedout = True

        if timedout:
            raise TimeoutException("Did not received frame in time (timeout=%s sec)" % timeout)

        return frame
```

lovezjh0909, 2022-07-12:
兄弟，我也碰到同样的问题，但我所有的服务都接收不到frame，但看总线信号发送和接收都有，不知道为啥。能否留个联系方式咨询下你。

mikisama, 2022-07-12:
Hi, @ziyangfu 

I've had the same issue.

Here is my solution. But it doesn't pass the unit test.

Accourding to ISO ISO 14229-2:2013, 7.4, key 4, when negative response code 0x78 is received, the client should restart its
 `P client` timer, using the enhanced reload value `P2*Client`.

```diff
diff --git a/udsoncan/client.py b/udsoncan/client.py
index 3181360..978044f 100755
--- a/udsoncan/client.py
+++ b/udsoncan/client.py
@@ -1936,7 +1936,7 @@ class Client:
             self.logger.debug("Waiting for server response")

             try:
-                if not respect_overall_timeout or (respect_overall_timeout and time.time() + single_request_timeout < overall_timeout_time):
+                if not respect_overall_timeout or (respect_overall_timeout and time.time() + single_request_timeout < overall_timeout_time) or using_p2_star:
                     timeout_type_used  = 'single_request'
                     timeout_value      = single_request_timeout
                 else:
```

pylessard, 2022-08-06:
Here's why. 
The library `request_timeout` is designed so that your code is sure to exit if the request does not complete after a certain time.
In your case, the device is responding with NRC 0x78 (RequestCorrectlyReceived_ResponsePending). That means that the client waits and try again, but the global timer keeps counting.  Since the connection read are blocking, the timeout of the blocking operation is adjusted at each subsequent call to guarantee that the global timeout is respected. So the overall time is 2sec, even though there was 3 attempts.

Like @mikisama said, if you don't want global timeout. set `client_config["request_timeout"] = None`

closed 2022-08-06 by pylessard

kongyun123, 2023-05-25:
> 兄弟，我也碰到同样的问题，但我所有的服务都接收不到frame，但看总线信号发送和接收都有，不知道为啥。能否留个联系方式咨询下你。

你好，解决了吗，我也遇到了这个问题，想咨询一下

jeffery96, 2024-05-16:
> 兄弟，我也碰到同样的问题，但我所有的服务都接收不到frame，但看总线信号发送和接收都有，不知道为啥。能否留个联系方式咨询下你。

兄弟，我也遇到同样的问题，请问你解决了嘛，能否交流下？

jeffery96, 2024-05-16:
> > 兄弟，我也碰到同样的问题，但我所有的服务都接收不到frame，但看总线信号发送和接收都有，不知道为啥。能否留个联系方式咨询下你。
> 
> 你好，解决了吗，我也遇到了这个问题，想咨询一下

兄弟，请问你解决了嘛？

kongyun123, 2024-05-16:
没解决，后来没用这个，完全手动收发报文实现的

lovezjh0909, 2024-05-24:
> > 兄弟，我也碰到同样的问题，但我所有的服务都接收不到frame，但看总线信号发送和接收都有，不知道为啥。能否留个联系方式咨询下你。
> 
> 你好，解决了吗，我也遇到了这个问题，想咨询一下
我这调通了，但其实也没搞明白是为啥，需要交流的话可以加我

lovezjh0909, 2024-05-24:
> > 兄弟，我也碰到同样的问题，但我所有的服务都接收不到frame，但看总线信号发送和接收都有，不知道为啥。能否留个联系方式咨询下你。
> 
> 你好，解决了吗，我也遇到了这个问题，想咨询一下

我这调通了，但也没搞明白原理，有需要可以加我w：18719267300