pylessard/python-udsoncan #135: can not write did

xixihaha66, 2022-08-23:
from can.interfaces.vector import VectorBus
from udsoncan.connections import PythonIsoTpConnection
from udsoncan.client import Client
import isotp
import udsoncan.configs
import time

# Refer to isotp documentation for full details about parameters
isotp_params = {
    'stmin': 32,
    # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
    'blocksize': 8,
    # Request the sender to send 8 consecutives frames before sending a new flow control message
    'wftmax': 0,  # Number of wait frame allowed before triggering an error
    'tx_data_length': 64,  # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
    'tx_data_min_length': 8,
    # Minimum length of CAN messages. When different from None, messages are padded to meet this length. Works with
    # CAN 2.0 and CAN FD.
    'tx_padding': 0xCC,  # Will pad all transmitted CAN messages with byte 0x00.
    'rx_flowcontrol_timeout': 1000,
    # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
    'rx_consecutive_frame_timeout': 1000,
    # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
    'squash_stmin_requirement': False,
    # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
    'max_frame_size': 4095,  # Limit the size of receive frame.
    'can_fd': False,
    'bitrate_switch': False
    # 'can_fd': True,
    # 'bitrate_switch': True
}

def myalgo(level, seed, params):
    """
    Builds the security key to unlock a security level. Returns the seed xor'ed with pre-shared key.
    """
    output_key = bytes()
    seed = int.from_bytes(seed, "big")
    seed = seed & 0xFFFFFFFF
    if seed != 0:
        for i in range(35):
            if (seed & 0x80000000) != 0:
                seed = seed ^ params
                seed = seed << 1
            else:
                seed = seed << 1
    seed = seed & 0xFFFFFFFF
    seed = seed.to_bytes(4, 'big')
    output_key = seed
    return bytes(output_key)

bus = VectorBus(channel=0, bitrate=500000)
tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x704,
                        rxid=0x784)  # Network layer addressing scheme
stack = isotp.CanStack(bus=bus, address=tp_addr,
                       params=isotp_params)  # Network/Transport layer (IsoTP protocol)
conn = PythonIsoTpConnection(stack)  # interface between Application and Transport layer
config = dict(udsoncan.configs.default_client_config)
config['data_identifiers'] = {
    0xF190: udsoncan.AsciiCodec(17)  # Codec that read ASCII string. We must tell the length of the string
}
config['use_server_timing'] = False
config['p2_timeout'] = 5
config['request_timeout'] = 5
config['use_server_timing'] = False
# config['security_algo'] = myalgo
# config['security_algo_params'] = dict(xorkey=b'\x12\x34\x56\x78')
with Client(conn, config=config) as client:                                     # Application layer (UDS protocol)
    client.change_session(3)
    # client.tester_present()
    response = client.request_seed(1, bytes(0))
    keycal = myalgo(1, response.service_data.seed, 0x4DBB0862)
    time.sleep(1)
    client.send_key(1, keycal)
    F190Value = 'XiaoPengD55.DFF.B'
    client.write_data_by_identifier(0xF190, F190Value)
    # response = client.read_data_by_identifier([0xF190])

C:\Users\XIAOPENG_BMS\AppData\Local\Programs\Python\Python37\python.exe D:/LHR/PyPrj/GG.py 
(c_ubyte_Array_8) <class 'IndexError'>: invalid index
[TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=5.000 sec)
Traceback (most recent call last):
  File "C:\Users\XIAOPENG_BMS\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\client.py", line 1946, in send_request
    payload = self.conn.wait_frame(timeout=timeout_value, exception=True)	
  File "C:\Users\XIAOPENG_BMS\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\connections.py", line 77, in wait_frame
    frame = self.specific_wait_frame(timeout=timeout)
  File "C:\Users\XIAOPENG_BMS\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\connections.py", line 499, in specific_wait_frame
    raise TimeoutException("Did not receive frame IsoTP Transport layer in time (timeout=%s sec)" % timeout)
udsoncan.exceptions.TimeoutException: Did not receive frame IsoTP Transport layer in time (timeout=5.0 sec)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:/LHR/PyPrj/GG.py", line 76, in <module>
    client.write_data_by_identifier(0xF190, F190Value)
  File "C:\Users\XIAOPENG_BMS\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\client.py", line 128, in decorated
    return func(self, *args, **kwargs)
  File "C:\Users\XIAOPENG_BMS\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\client.py", line 439, in write_data_by_identifier
    response = self.send_request(req)
  File "C:\Users\XIAOPENG_BMS\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\client.py", line 1954, in send_request
    raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' % (timeout_name_to_report, timeout_value))
udsoncan.exceptions.TimeoutException: Did not receive response in time. Global request timeout time has expired (timeout=5.000 sec)

Process finished with exit code 1

xixihaha66, 2022-08-23:
> from can.interfaces.vector import VectorBus from udsoncan.connections import PythonIsoTpConnection from udsoncan.client import Client import isotp import udsoncan.configs import time
> 
> isotp_params = { 'stmin': 32, # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9 'blocksize': 8, # Request the sender to send 8 consecutives frames before sending a new flow control message 'wftmax': 0, # Number of wait frame allowed before triggering an error 'tx_data_length': 64, # Link layer (CAN layer) works with 8 byte payload (CAN 2.0) 'tx_data_min_length': 8, # Minimum length of CAN messages. When different from None, messages are padded to meet this length. Works with # CAN 2.0 and CAN FD. 'tx_padding': 0xCC, # Will pad all transmitted CAN messages with byte 0x00. 'rx_flowcontrol_timeout': 1000, # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds 'rx_consecutive_frame_timeout': 1000, # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds 'squash_stmin_requirement': False, # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible. 'max_frame_size': 4095, # Limit the size of receive frame. 'can_fd': False, 'bitrate_switch': False # 'can_fd': True, # 'bitrate_switch': True }
> 
> def myalgo(level, seed, params): """ Builds the security key to unlock a security level. Returns the seed xor'ed with pre-shared key. """ output_key = bytes() seed = int.from_bytes(seed, "big") seed = seed & 0xFFFFFFFF if seed != 0: for i in range(35): if (seed & 0x66000000) != 0: seed = seed ^ params seed = seed << 1 else: seed = seed << 1 seed = seed & 0xFFFFFFFF seed = seed.to_bytes(4, 'big') output_key = seed return bytes(output_key)
> 
> bus = VectorBus(channel=0, bitrate=500000) tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x704, rxid=0x784) # Network layer addressing scheme stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params) # Network/Transport layer (IsoTP protocol) conn = PythonIsoTpConnection(stack) # interface between Application and Transport layer config = dict(udsoncan.configs.default_client_config) config['data_identifiers'] = { 0xF190: udsoncan.AsciiCodec(17) # Codec that read ASCII string. We must tell the length of the string } config['use_server_timing'] = False config['p2_timeout'] = 5 config['request_timeout'] = 5 config['use_server_timing'] = False with Client(conn, config=config) as client: # Application layer (UDS protocol) client.change_session(3) # client.tester_present() response = client.request_seed(1, bytes(0)) keycal = myalgo(1, response.service_data.seed, 0x11111111) time.sleep(1) client.send_key(1, keycal) F190Value = 'ABCDABCDABCDABCDA' client.write_data_by_identifier(0xF190, F190Value) # response = client.read_data_by_identifier([0xF190])
> 
> C:\Users\AppData\Local\Programs\Python\Python37\python.exe D:/LHR/PyPrj/GG.py (c_ubyte_Array_8) <class 'IndexError'>: invalid index [TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=5.000 sec) Traceback (most recent call last): File "C:\Users\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\client.py", line 1946, in send_request payload = self.conn.wait_frame(timeout=timeout_value, exception=True) File "C:\Users\XIAOPENG_BMS\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\connections.py", line 77, in wait_frame frame = self.specific_wait_frame(timeout=timeout) File "C:\Users\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\connections.py", line 499, in specific_wait_frame raise TimeoutException("Did not receive frame IsoTP Transport layer in time (timeout=%s sec)" % timeout) udsoncan.exceptions.TimeoutException: Did not receive frame IsoTP Transport layer in time (timeout=5.0 sec)
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last): File "D:/LHR/PyPrj/GG.py", line 76, in client.write_data_by_identifier(0xF190, F190Value) File "C:\Users\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\client.py", line 128, in decorated return func(self, *args, **kwargs) File "C:\Users\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\client.py", line 439, in write_data_by_identifier response = self.send_request(req) File "C:\Users\AppData\Local\Programs\Python\Python37\lib\site-packages\udsoncan\client.py", line 1954, in send_request raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' % (timeout_name_to_report, timeout_value)) udsoncan.exceptions.TimeoutException: Did not receive response in time. Global request timeout time has expired (timeout=5.000 sec)
> 
> Process finished with exit code 1 [image]

why 2E service did't work, f190  data didn't sent

rajagopalvn, 2022-08-26:
Hi pylessard, I implemented UDS in application code, i need to test UDS.
How to run python-udsoncan script for UDS testing ?

Thanks.
Aajai

xixihaha66, 2022-08-26:
>

closed 2022-08-26 by xixihaha66