pylessard/python-udsoncan #136: How to run python-udsoncan

rajagopalvn, 2022-08-26:
Hi, I have implemented UDS services in application code, and need to test all UDS service id.
I am planning to use your python(python-udsoncan) script for UDS testing.
how to run your python-udsoncan script for UDS testing.

Thanks in advance.
Aajai.

pylessard, 2022-08-26:
Care to be more specific with your questions?
I wrote documentation to explain how to use the project. Maybe start by reading it?

Aajaivn, 2022-08-29:
Hello, I read documentation, but still not clear which python file should run for UDS service testing ?
Please can you point out in document.

pylessard, 2022-08-29:
Hi, I will put as much effort in my answers as you put effort in writing your questions. 
Right now you made none, so I making none.

Be more specific, ask an actual questions. Show what you did. Tell what you don't understand. Then maybe I can answer something.

Kaeptn-G, 2022-09-24:
> Hi, I have implemented UDS services in application code, and need to test all UDS service id. I am planning to use your python(python-udsoncan) script for UDS testing. how to run your python-udsoncan script for UDS testing?
> 
> Thanks in advance. Aajai.

Mate, I think you misunderstood what python-udsoncan actually is. Seema like you are searching for a ready-to-run application, but this is more a library which helps you to habdle all the protocol stuff. The app using this library, needs to be created by you. 
Was this what you were missing?

pylessard, 2022-09-25:
I will close this issue as it is empty in content.

closed 2022-09-25 by pylessard