pylessard/python-udsoncan #140: udsoncan.MemoryLocation support for 64-bit address space

driftregion, 2022-12-27:
Hi @pylessard , thank you for python-udsoncan.  I have a question about `MemoryLocation`.

https://github.com/pylessard/python-udsoncan/blob/85fdb4e8f2b90adf1b9604b93cab78fa21ff237f/udsoncan/__init__.py#L297 

The specific use-case I'd like to support is making requests to a server with a 64-bit address space, e.g. `address_format=64`. I don't have a copy of the 2006 standard, but Annex H in the 2020 standard seems to indicate that `addressAndLengthFormatIdentifier` > `0x45` is valid, for example: `0x88`

I'm happy to submit a pull request that adds tests and support for `addressAndLengthFormatIdentifier` > `0x45`, but I'd like to first know whether this conflicts with the direction of python-udsoncan.  Specifically:

- are there 2006 clients that rely on `MemoryLocation` throwing a `ValueError` for `addressAndLengthFormatIdentifier` > `0x45` or can this ValueError be safely removed? 
- is the specification of `addressAndLengthFormatIdentifier` different between standard versions? (does `MemoryLocation` need to check `client.config['standard_version']`?)

Thanks!

pylessard, 2022-12-27:
The standard is not so clear about that honnestly. The examples stops at the values that are supported in the library.

[image]

I think there is no downside to supporting 64 bits addresses. Can you tell me if branch [support-alfid-64bits](https://github.com/pylessard/python-udsoncan/tree/support-alfid-64bits)  is enough for you?

driftregion, 2022-12-28:
I tried branch [support-alfid-64bits](https://github.com/pylessard/python-udsoncan/tree/support-alfid-64bits).  The `autosize_address` and `autosize_memorysize` functions need updating also.

```sh
  File "/home/user/repos/python-udsoncan/udsoncan/__init__.py", line 375, in __init__
    address_format = self.autosize_address(address)
  File "/home/user/repos/python-udsoncan/udsoncan/__init__.py", line 408, in autosize_address
    raise ValueError("address size must be smaller or equal than 40 bits")
ValueError: address size must be smaller or equal than 40 bits
```

 After updating those functions, it works fine in my case.

[patch.txt](https://github.com/pylessard/python-udsoncan/files/10311264/patch.txt)

pylessard, 2022-12-28:
Updated. Will merge once you confirm

driftregion, 2022-12-28:
Confirmed working.

pylessard, 2023-01-09:
Merged in #142

closed 2023-01-09 by pylessard