pylessard/python-udsoncan #145: Issue with SecurityAccess

pompushko, 2023-01-28:


pylessard, 2023-01-28:
1st byte is an echo of the security access, 4 last bytes is the seed.  SecurityAccess.interpret_response will extract it and put it in response.response_data, check yhe doc 
 https://udsoncan.readthedocs.io/en/latest/udsoncan/services.html#udsoncan.services.SecurityAccess.interpret_response

Also, do you know that the client can unlock a session by itself? You just need to provide your algorithm. Check the doc again

https://udsoncan.readthedocs.io/en/latest/udsoncan/services.html#udsoncan.services.SecurityAccess.interpret_response

pompushko, 2023-01-28:
> 1st byte is an echo of the security access, 4 last bytes is the seed. SecurityAccess.interpret_response will extract it and put it in response.response_data, check the doc https://udsoncan.readthedocs.io/en/latest/udsoncan/services.html#udsoncan.services.SecurityAccess.interpret_response

Thank you for a fast response :)
But I still cannot understand, how to make integer from that bytes..
Could you, please, help me?
`seedint = int.from_bytes(response.data, "big")`

pylessard, 2023-01-28:
I'm pretty sure you haven't looked much at the link I gave you :)

So what you are trying to do is : `int.from_bytes(response.data[1:4])`   this takes bytes 1,2,3,4, skipping bytes #0.
But this method is really not the best approach.  The library can do that for you.

A cleaner approach would be:
```
SecurityAccess.interpret_response(response, SecrutiyAccess.Mode.RequestSeed)
int.from_bytes(response.service_data.seed)
```

But even that is unnecessary work. You could replace most of your code by something like this: 

```
def myalgo(level, seed, params):
    # level = 5 in this case. Params=None because security_algo_params is unset
    seedint = int.from_bytes(seed, 'big')
    key = get_key_by_seed(seedint)
    keybytes = key.to_bytes(4, byteorder="big")
    return keybytes

client.set_config('security_algo', my_algo)
client.unlock_security_access(5)   # Everything happens here. Exception in case of failure.

```

pompushko, 2023-01-28:
Yeah, I put response.data[1:] it start works fine.
Thank you.

closed 2023-01-28 by pompushko