pylessard/python-udsoncan #148: Getting Raw payload

Aziz-repo, 2023-03-08:
I'm trying to read DTC from an ECU and I'm getting a positive Response. When I print the response it just prints the name of the class.
My Question is: How can I get the raw data from the response object ?
I checked the `from_payload()` and `get_payload()` but I don't think it's the right approach to get it done.

pylessard, 2023-03-08:
Hi. 
DTC parsing works but lacks documentation.

`response.data` gives the raw payload.
If you used the client object, the payload will be parsed in  `response.service_data`.

You can use IPython to inspect the content of the data. Or just print the object with `print(response service_data.__dict__`

Finally, the unit tests are a good source of examples.

Cheers

Aziz-repo, 2023-03-08:
Thank you for the fast reply!!

As I understand the source code, there is a nested class called `ResponseData` and it's attributes are specific to each service.
I think it's well documented, I just stumped on it after opening the thread but thank you for the advice :blush:

pylessard, 2023-03-09:
Shoul we keep this thread open?

Aziz-repo, 2023-03-09:
Yes,
I'm closing this thread for no issue to be solved.

closed 2023-03-09 by Aziz-repo