pylessard/python-udsoncan #151: Docs changes

Aziz-repo, 2023-03-15:
Hi,
This period I'm working on a project involving a ECU in an automotive industry and I frequently check the docs presented in [https://udsoncan.readthedocs.io/en/stable/index.html](url) .
I checked the docs today and I noticed that some of the docs was not to be found. (Particularly in the services and client sections) 
I noticed also that a merge request was accepted and the master branch was updated but even if when I checked the changes the docs where not changed.

pylessard, 2023-03-15:
You are right, something is broken on readthedocs.

Let me see

pylessard, 2023-03-15:
Ok. So I released V1.17 yesterday.
The big change in 1.17 is that I introduced type hints.  I used the `TypedDict` class at few places that is not supported by python 3.7 and readthedocs.org uses python 3.7.   The build log was full of ImportError.

I added a stub for TypedDict for python versions below 3.8. It fixes the docs and also fixes the backward compatibility issue that came iwth TypeDict (I think). 

Will release 1.17.1 with that fix. 
Thank you

pylessard, 2023-03-15:
I think it is all good. 
Fixed with V1.17.1

Cheers

closed 2023-03-15 by pylessard

Aziz-repo, 2023-03-15:
Thank you for the hard work 😊