pylessard/python-udsoncan #157: Request/Response.from_payload(...) doesn't find IDs any more

trd1st, 2023-05-23:
Hi pylessard,

in the latest releases (1.17 and 1.17.1), the from_payload(...) function from Request and Response does not seem to resolve CAN-IDs any more.
The cause seems to be, that BaseService.__get_all_subclasses(cls) in BaseService.from_request_id(cls, given_id) seems to return an empty array.

I have attached a minimal working example, which works correctly in 1.16, but not in 1.17 and 1.17.1.

```
from udsoncan import Request, Response

def _extract_service_id_name(
    uds_payload = None
):
    if uds_payload is None:
        return None
    try:
        try:
            uds = Request.from_payload(uds_payload)
            service_id_string = f"{uds.service.__name__}"
        except AttributeError:
            uds = Response.from_payload(uds_payload)
            service_id_string = f"{uds.service.__name__ }: {uds.code_name}"
    except AttributeError:
        service_id_string = None
    return service_id_string

if __name__ == "__main__":
    extracted_service_id_name = _extract_service_id_name(bytearray([0x10, 0x02]))
    print(f"Service-ID-name: {extracted_service_id_name}")
    extracted_service_id_name = _extract_service_id_name(bytearray([0x50, 0x02]))
    print(f"Service-ID-name: {extracted_service_id_name}")
```

Is this an issue, or do i have to change something on my side for 1.17+?

I hope you have a nice day,
Daniel

pylessard, 2023-05-24:
That's a bug. 
The problem is the Response object that can't get the code name. It's a 1 line fix. Will publish 1.17.2 soonand fix that

pylessard, 2023-05-24:
You are right. I see the error.
I have moved BaseService out of the service folder because of a circular dependency problem. 
__get_all_subclasses will work only if the service module is loaded. 

Of course, when the unit test runs, this module gets loaded before the test_response ad the test_request test suite.

Will fix

pylessard, 2023-05-24:
As a workaround, you can replace your first line by `from udsoncan import Request, Response, services`

trd1st, 2023-05-25:
Thanks for the fast reply and resolution.
I approved your pr, as your change fixes the problem on my side  :)
Thanks for the workaround code, but I already excluded 1.17.0 and 1.17.1 from my pipeline.
No problem from my side to stay at 1.16 until 1.17.2 or 1.18 goes live :)

closed 2023-05-25 by pylessard

pylessard, 2023-05-25:
Fixed in [v1.17.2](https://github.com/pylessard/python-udsoncan/releases/tag/v1.17.2)
Cheers

trd1st, 2023-05-30:
Thanks a lot for the fast fix :)

commit 272569ef (2023-05-25): Parse request/response from payload success depends on module loading order (#158)

* Fixed #157 : Parsing payload may fail depending on module loading order