pylessard/python-udsoncan #159: Connection error after Reset

Aziz-repo, 2023-05-30:
Hey,
This is just a question rather than an issue.
I'm trying the ecu reset service and it works fine but when I try to reconnect to the ECU it fails with 2 possible errors:

1. [Errno 104] Connection reset by peer
2. [Errno 32 ] Broken pipe

the code where I implement this functionnality is:
```python
with self.uds_client as client:
    response = client.ecu_reset(type)
    client.close()
return response
```
Is that the right way to implement that??

pylessard, 2023-05-30:
hmm, I do not know much about the DoIP connection. But I see few things wrong here.

When you use the `with` statement, the client is closed at the end, so no need to close it.
When the client clsoes, it also closes the connection. Probably your issue.

I see that your client already exists, why no use it directly.  There's no need to close the client after every request.  You open once at the beginning of your program and close at exit.

```python
return self.uds_client.ecu_reset
```

Aziz-repo, 2023-05-31:
I solved the issue by removing the `client.close()` and setting the flag of `auto_reconnect_tcp` in the doip object to True.
It was a transport layer issue (DoIP in my case) because it's responsible for the opening and closing the tcp connection.(Quite logical I don't know how I missed it :sweat_smile: )
I'm closing this issue, Thank you for the fast reply!!

closed 2023-05-31 by Aziz-repo