pylessard/python-udsoncan #162: Get Subfunction from Response object

Aziz-repo, 2023-06-23:
Hey,
Is there a way of retrieving the subfunction used in the request from the response object?

pylessard, 2023-06-23:
You mean when using the client?
Right now, you can't. Only the subfunction read from the response.

pylessard, 2023-06-23:
Do you have a good use-case for that?

Aziz-repo, 2023-06-23:
well now I'm using the client to get data from an ECU and I'm in the middle of getting responses concerning DTCs and I want to interpret the responses to fill the service_data attribute. As you know `interpret_response()` method in the ReadDTCInformation class need a subfunction and I want to get it dynamically.

pylessard, 2023-06-23:
Not sure if I understand correctly, but the client do call `interpret_response` for you, so the service_data attribute is filled.

If you manage Services manually, then you should have your request available.

Am I missing something?

Aziz-repo, 2023-06-23:
In other word I sending requests using the client then I use the `BaseService.interpret_response()` where I substitute the BaseService with the actual Service name. I did it this way because I'm developing a GUI application where the user can configure requests using dropboxes and receive responses and the data within. That data must be displayed in details and I didn't find another way to achieve that other than the previously mentioned way

pylessard, 2023-06-23:
debugging over dropbox is definetly not a use case I predicted! ha ha.

If the client attach the request to the response inside a property called "original_request", would that help?

Aziz-repo, 2023-06-23:
That will be awesome. I can get a lot of data from that attribute if it's provided and only rely on the response to get the data.
Regarding the GUI, I'm going for something like vector testing software if you are familiar with it but it is more user friendly specially for testers out of scope of automotive industry.

pylessard, 2023-06-23:
I can add this easily tonight. Expect a release during the weekend.
Cheers

pylessard, 2023-06-24:
Implemented in v1.18.1. Please confirm that this resolve your problem.
Thank you!

Aziz-repo, 2023-06-30:
Hey again,
Sorry for the late response I've tested the new attribute and it seems stable. It served well the use case in hand. Thank you!

commit 391e5efe (2023-06-24): Attach request to response in client (#163)

* Fix #162 Attach original request to response in client

* fix #161 - Bad static type in send_request
closed 2023-06-30 by Aziz-repo