pylessard/python-udsoncan #164: support 0x29

zhaohuijun1, 2023-07-11:
Hi，When support 0x29 Services?

zhaohuijun1, 2023-07-12:
OK,Thanks.

zhaohuijun1, 2023-08-03:
Hi，do you have any plans for 0x29 services?

geynis, 2023-09-11:
I hope I can contribute a PR in the near future. Please tell me if someone else is already working on it.

pylessard, 2023-09-11:
Not to my knowledge. If you do, please folliw the module architecture and consider unit tests and documentation.

Cheers

geynis, 2023-09-12:
Sure, already started, thanks!

geynis, 2023-09-13:
Hi, It would be great if you can feedback on it, core development done (was easier than expected). 
Next step, I need to write the unit tests, here is the branch: 
[authentication_service_0x29 ](https://github.com/geynis/python-udsoncan/tree/authentication_service_0x29 )

pylessard, 2023-09-18:
@geynis Nice!
I have checked the code you wrote, seems super clean and you clearly followed the project architecture.
I have only checked the container, not the content. Meaning, I have not validated the compliance with ISO-14229:2020.  I will 
put my trust on you for this.

The docstring are nice, don't forget to add an entry in the doc folder to make sphinx load them.

Do you support all of the possible flows proposed by the standard? If not, maybe make sure what is supported and what is not is clearly identified.

geynis, 2023-09-20:
Hi, thanks for the feedback. I think I have now added the entries you mentioned in the doc folder.
It is supporting all possible flows of the service, it has 9 sub functions and they are all there. I have added unit-tests.
My next step is to test with a real setup, but it will be anyway partial as there are many possibilities and I don't have all examples in a real setup (and don't have the private keys to reach end-to-end testing)

geynis, 2023-09-26:
Submitted PR

pylessard, 2023-09-27:
Merged  and part of new release 1.20.0

closed 2023-09-27 by pylessard

geynis, 2024-11-02:
@pylessard I mentioned this repo in my blog post about this new UDS service:
[Link](https://embeddedcomputing.com/application/automotive/new-uds-authentication-enhanced-security-familiar-challenges)