pylessard/python-udsoncan #168: 1904 Error returned while reading DTC snapshot

hanlfa, 2023-08-07:
2023-08-07 12:07:53 [INFO] UdsClient: ReadDTCInformation<0x19> - Sending request with subfunction "reportDTCSnapshotRecordByDTCNumber" (0x04).
2023-08-07 12:07:53 [DEBUG] Connection: Sending 6 bytes : [b'1904c25900ff']
2023-08-07 12:07:54 [DEBUG] Connection: Received 97 bytes : [b'5904c25900090119cf0078cf010000cf02170404000000cf03000000ce000000ce010019ce02006cce0300ce040026ce0500ce0641ce0741ce0841ce0943ce0a43ce0b43ce0c00ce0d2364ce0e23ce0f00ce1001ce1102ce1201ce1303ce140000']
2023-08-07 12:07:54 [INFO] UdsClient: Received positive response for service ReadDTCInformation (0x19) from server.
2023-08-07 12:07:54 [ERROR] UdsClient: [ConfigError] : Actual data identifier configuration contains no definition for data identifier 0xcf00
2023-08-07 12:07:54 [INFO] Connection: Connection closed
As per the above log, unable to output response. How to solve this problem

pylessard, 2023-08-08:
As the message say, you need to define a codec to interpret the data you have received.  See example here : 
https://udsoncan.readthedocs.io/en/latest/udsoncan/examples.html#reading-a-did-with-readdatabyidentifier

The service/subfunction you are using returns data for DID 0xCF00, so you need that in your config

hanlfa, 2023-08-09:
ok,it is right.
i have a question,
How to read a snapshot while reading a dtc? Must 1902 read dtc before 1904 read snapshot? Can you read the dtc and the corresponding snapshot at once

pylessard, 2023-08-09:
If by "reading a DTC" you mean getting its status, you can use 0x1904 only. It returns the satus byte + snapshot data in a single call.

pylessard, 2023-08-18:
Should we close this?

hanlfa, 2023-08-18:
ok

closed 2023-09-19 by pylessard