pylessard/python-udsoncan #171: the logging module and can data are inconsistent

hanlfa, 2023-09-07:
logging:
2023-09-07 16:03:02 [DEBUG] Connection: Sending 2 bytes : [b'1101']
2023-09-07 16:03:02 [DEBUG] Connection: Received 3 bytes : [b'7f1178']
2023-09-07 16:03:04 [DEBUG] Connection: Received 3 bytes : [b'7f1178']
2023-09-07 16:03:05 [DEBUG] Connection: Received 2 bytes : [b'5101']
2023-09-07 16:03:05 [INFO] UdsClient: Received positive response for service ECUReset (0x11) from server.
-->Reset ECU1101 is ok
2023-09-07 16:03:15 [INFO] UdsClient: ReadDataByIdentifier<0x22> - Reading data identifier : 0xf195 (SystemSupplierECUSoftwareVersionNumberDataIdentifier)
2023-09-07 16:03:15 [DEBUG] Connection: Sending 3 bytes : [b'22f195']
2023-09-07 16:03:15 [DEBUG] Connection: Received 11 bytes : [b'62f19556335f00015f0003']
2023-09-07 16:03:15 [INFO] UdsClient: Received positive response for service ReadDataByIdentifier (0x22) from server.
SW: V3_0001_0003
F195: V3_0001_0003
2023-09-07 16:03:18 [INFO] Connection: Connection opened
2023-09-07 16:03:18 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Switching session to extendedDiagnosticSession (0x03)
2023-09-07 16:03:18 [DEBUG] Connection: Sending 2 bytes : [b'1003']
2023-09-07 16:03:33 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame IsoTP Transport layer in time (timeout=15 sec) 
2023-09-07 16:03:33 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=15.000 sec)
2023-09-07 16:03:33 [INFO] UdsClient: CommunicationControl<0x28> - ControlType=0x00 (enableRxAndTx) - Sending request with a CommunicationType byte of 0x03 (subnet=0x0. Flags : [NormalMsg,NetworkManagementMsg])
2023-09-07 16:03:33 [DEBUG] Connection: Sending 3 bytes : [b'280003']
2023-09-07 16:03:48 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame IsoTP Transport layer in time (timeout=15 sec) 
2023-09-07 16:03:48 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=15.000 sec)
Tester turn on communication failed.
2023-09-07 16:03:48 [INFO] UdsClient: ControlDTCSetting<0x85> - SettingType=0x01 (on) - Turning DTC On with a payload of 0 bytes
2023-09-07 16:03:48 [DEBUG] Connection: Sending 2 bytes : [b'8501']
2023-09-07 16:04:03 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame IsoTP Transport layer in time (timeout=15 sec) 
2023-09-07 16:04:03 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=15.000 sec)
Tester turn on DTC failed.Did not receive response in time. P2 timeout time has expired (timeout=15.000 sec)
2023-09-07 16:04:03 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Switching session to defaultSession (0x01)
2023-09-07 16:04:03 [DEBUG] Connection: Sending 2 bytes : [b'1001']
2023-09-07 16:04:03 [DEBUG] Connection: Received 6 bytes : [b'5001003201f4']
2023-09-07 16:04:03 [INFO] UdsClient: Received positive response for service DiagnosticSessionControl (0x10) from server.
-->enter 1001 ok
2023-09-07 16:04:03 [INFO] Connection: Connection closed
2023-09-07 16:04:03 [INFO] Connection: Connection closed
2023-09-07 16:04:03 [INFO] Connection: Connection opened
2023-09-07 16:04:03 [INFO] UdsClient: ClearDiagnosticInformation<0x14> - Clearing all DTCs (group mask : 0xFFFFFF)
2023-09-07 16:04:03 [DEBUG] Connection: Sending 4 bytes : [b'14ffffff']
2023-09-07 16:04:03 [DEBUG] Connection: Received 3 bytes : [b'7f1478']
2023-09-07 16:04:03 [DEBUG] Connection: Received 1 bytes : [b'54']
2023-09-07 16:04:03 [INFO] UdsClient: Received positive response for service ClearDiagnosticInformation (0x14) from server.
-->Firmware updated successfully!
2023-09-07 16:04:03 [INFO] Connection: Connection closed

Can data:
[image]

hanlfa, 2023-09-07:
The log module return an timeout error,but the actual can data is normal,all requests have returned a positive reponse.

hanlfa, 2023-09-07:
[image]

hanlfa, 2023-09-10:
Because I need to print some information according to the response data, the return timeout error gets no response and cannot make further judgment.
who can help me?

pylessard, 2023-09-18:
Hi, you should share some code.
Either your ids are not configured properly or you are using the `isotp.CanStack` and you consume all the messages in parallel.

hanlfa, 2023-09-19:
this is code,The code was simplified due to confidentiality system, but the timeout problem persisted。

import os
import time
import bincopy
import threading
from crc import Configuration,Calculator
import hashlib
import re
import can
import argparse
import datetime
import isotp
import udsoncan
from can.interfaces.pcan import PcanBus
from udsoncan import Routine, services
from udsoncan.connections import PythonIsoTpConnection
from udsoncan.client import Client
from udsoncan.exceptions import *
from udsoncan.services import *
from udsoncan import DidCodec

class F195_did(udsoncan.DidCodec):
   def decode(self, payload):
      stage_version=chr(payload[0])+ chr(payload[1]) #默认10进制，需要根据实际转化
      xiahuaxian_1=chr(payload[2])  #_
      MajorVersion=str(payload[3]).zfill(2)+str(payload[4]).zfill(2)
      xiahuaxian_2 = chr(payload[5])  # _
      MinorVersion = str(payload[6]).zfill(2)+str(payload[7]).zfill(2)
      sw=stage_version+xiahuaxian_1+MajorVersion+xiahuaxian_2+MinorVersion
      print("SW:",sw)
      return sw
   def __len__(self):
      return 8    # encoded payload is 4 byte long

class StubbedDidCodec(DidCodec):
    def encode(self, did_value):
        return   did_value

    def decode(self, did_payload):
        return   did_payload

    def __len__(self):
        return 1

config = dict(udsoncan.configs.default_client_config)
config['use_server_timing'] = False
config['request_timeout'] = 30
config['p2_timeout'] =  0.05   #1
config['request_timeout'] = 25
config['p2_star_timeout'] =  5
config['data_identifiers'] = {
   0xF195 :F195_did,
}

tester_present_flag=False

def simple_periodic_send():
   """
   #发送一条周期为4000ms的周期报文到总线
   """
   while tester_present_flag:
      msg = can.Message(arbitration_id=0x7DF, data=[0x02, 0x3E, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00], is_extended_id=False)
      #bus.send_periodic(msg, 4)
      bus.send(msg)
      time.sleep(4)
def uds_download():
   diag_funcreq = 0x7DF
   diag_phyreq = 0x61B
   diag_phyresp = 0x71B
[... 155 lines cut ...]
      'rx_consecutive_frame_timeout': 50,# Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds,N_CS
      'squash_stmin_requirement': False,
      # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
      'max_frame_size': 4095,  # Limit the size of receive frame.
      'can_fd':False
   }
   try:
      bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=500000)
   except:
      print('Connect PCAN failed.')
      os.system('pause')
      os._exit(0)

   print('ECU OTA PC Tool V1.0')
   thread1 = threading.Thread(target=simple_periodic_send)
   thread2 = threading.Thread(target=uds_download )
   # Enable to logging output
   udsoncan.setup_logging()
   thread2.start()
   thread1.start()
   os.system('pause')
   # 等待线程结束
   os._exit(0)
   thread1.join()
   thread2.join()

hanlfa, 2023-09-19:
this is udsoncan log
2023-09-18 17:30:29 [INFO] Connection: Connection opened
2023-09-18 17:30:29 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Switching session to extendedDiagnosticSession (0x03)
2023-09-18 17:30:29 [DEBUG] Connection: Sending 2 bytes : [b'1003']
2023-09-18 17:30:29 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame IsoTP Transport layer in time (timeout=0.05 sec) 
2023-09-18 17:30:29 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=0.050 sec)
2023-09-18 17:30:29 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Switching session to extendedDiagnosticSession (0x03)
2023-09-18 17:30:29 [DEBUG] Connection: Sending 2 bytes : [b'1003']
2023-09-18 17:30:29 [DEBUG] Connection: Received 6 bytes : [b'5003003201f4']
2023-09-18 17:30:29 [INFO] UdsClient: Received positive response for service DiagnosticSessionControl (0x10) from server.
2023-09-18 17:30:29 [INFO] Connection: Connection closed
2023-09-18 17:30:29 [INFO] Connection: Connection opened
2023-09-18 17:30:29 [INFO] UdsClient: RoutineControl<0x31> - ControlType=0x01 - Starting routine ID 0x0203 (VehicleManufacturerSpecific) with a payload of 0 bytes
2023-09-18 17:30:29 [DEBUG] Connection: Sending 4 bytes : [b'31010203']
2023-09-18 17:30:29 [DEBUG] Connection: Received 5 bytes : [b'7101020300']
2023-09-18 17:30:29 [INFO] UdsClient: Received positive response for service RoutineControl (0x31) from server.
2023-09-18 17:30:29 [INFO] Connection: Connection closed
2023-09-18 17:30:29 [INFO] Connection: Connection opened
2023-09-18 17:30:29 [INFO] UdsClient: ControlDTCSetting<0x85> - SettingType=0x02 (off) - Turning DTC Off with a payload of 0 bytes
2023-09-18 17:30:29 [DEBUG] Connection: Sending 2 bytes : [b'8502']
2023-09-18 17:30:29 [DEBUG] Connection: Received 2 bytes : [b'c502']
2023-09-18 17:30:29 [INFO] UdsClient: Received positive response for service ControlDTCSetting (0x85) from server.
2023-09-18 17:30:29 [INFO] UdsClient: CommunicationControl<0x28> - ControlType=0x03 (disableRxAndTx) - Sending request with a CommunicationType byte of 0x03 (subnet=0x0. Flags : [NormalMsg,NetworkManagementMsg])
2023-09-18 17:30:29 [DEBUG] Connection: Sending 3 bytes : [b'280303']
2023-09-18 17:30:29 [DEBUG] Connection: Received 2 bytes : [b'6803']
2023-09-18 17:30:29 [INFO] UdsClient: Received positive response for service CommunicationControl (0x28) from server.
2023-09-18 17:30:29 [INFO] Connection: Connection closed
2023-09-18 17:30:29 [INFO] Connection: Connection opened
2023-09-18 17:30:29 [INFO] UdsClient: ReadDataByIdentifier<0x22> - Reading data identifier : 0xf195 (SystemSupplierECUSoftwareVersionNumberDataIdentifier)
2023-09-18 17:30:29 [DEBUG] Connection: Sending 3 bytes : [b'22f195']
2023-09-18 17:30:30 [DEBUG] Connection: Received 11 bytes : [b'62f19556345f00015f0001']
2023-09-18 17:30:30 [INFO] UdsClient: Received positive response for service ReadDataByIdentifier (0x22) from server.
2023-09-18 17:30:30 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Switching session to programmingSession (0x02)
2023-09-18 17:30:30 [DEBUG] Connection: Sending 2 bytes : [b'1002']
2023-09-18 17:30:30 [DEBUG] Connection: Received 3 bytes : [b'7f1078']
2023-09-18 17:30:30 [DEBUG] Connection: Received 6 bytes : [b'5002003201f4']
2023-09-18 17:30:30 [INFO] UdsClient: Received positive response for service DiagnosticSessionControl (0x10) from server.
2023-09-18 17:30:30 [INFO] UdsClient: ECUReset<0x11> - Requesting reset of type 0x01 (hardReset)
2023-09-18 17:30:30 [DEBUG] Connection: Sending 2 bytes : [b'1101']
2023-09-18 17:30:30 [DEBUG] Connection: Received 3 bytes : [b'7f1178']
2023-09-18 17:30:32 [DEBUG] Connection: Received 3 bytes : [b'7f1178']
2023-09-18 17:30:33 [DEBUG] Connection: Received 2 bytes : [b'5101']
2023-09-18 17:30:33 [INFO] UdsClient: Received positive response for service ECUReset (0x11) from server.
2023-09-18 17:30:43 [INFO] UdsClient: ReadDataByIdentifier<0x22> - Reading data identifier : 0xf195 (SystemSupplierECUSoftwareVersionNumberDataIdentifier)
2023-09-18 17:30:43 [DEBUG] Connection: Sending 3 bytes : [b'22f195']
2023-09-18 17:30:43 [DEBUG] Connection: Received 11 bytes : [b'62f19556345f00015f0001']
2023-09-18 17:30:43 [INFO] UdsClient: Received positive response for service ReadDataByIdentifier (0x22) from server.
2023-09-18 17:30:46 [INFO] Connection: Connection opened
2023-09-18 17:30:46 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Switching session to extendedDiagnosticSession (0x03)
2023-09-18 17:30:46 [DEBUG] Connection: Sending 2 bytes : [b'1003']
2023-09-18 17:31:01 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame IsoTP Transport layer in time (timeout=15 sec) 
2023-09-18 17:31:01 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=15.000 sec)
2023-09-18 17:31:01 [INFO] UdsClient: CommunicationControl<0x28> - ControlType=0x00 (enableRxAndTx) - Sending request with a CommunicationType byte of 0x03 (subnet=0x0. Flags : [NormalMsg,NetworkManagementMsg])
2023-09-18 17:31:01 [DEBUG] Connection: Sending 3 bytes : [b'280003']
2023-09-18 17:31:01 [DEBUG] Connection: Received 2 bytes : [b'6800']
2023-09-18 17:31:01 [INFO] UdsClient: Received positive response for service CommunicationControl (0x28) from server.
[... cut ...]
2023-09-18 17:31:31 [INFO] UdsClient: ClearDiagnosticInformation<0x14> - Clearing all DTCs (group mask : 0xFFFFFF)
2023-09-18 17:31:31 [DEBUG] Connection: Sending 4 bytes : [b'14ffffff']
2023-09-18 17:31:31 [DEBUG] Connection: Received 3 bytes : [b'7f1478']
2023-09-18 17:31:31 [DEBUG] Connection: Received 1 bytes : [b'54']
2023-09-18 17:31:31 [INFO] UdsClient: Received positive response for service ClearDiagnosticInformation (0x14) from server.
2023-09-18 17:31:31 [INFO] Connection: Connection closed

pylessard, 2023-09-19:
Thanks.
First, you don't need to call `DiagnosticSessionControl.interpret_response(response)`. The client does it for you.

Also I find weird that you create and destroy a client for each request. You can use the same client.

In the canalyzer log that you shared, I do not see the message that is missing, so I see no issue so far.  
You can enable the "isotp" logger in debug mode as well if you want to see more.

hanlfa, 2023-09-20:
1.ok,i will delete DiagnosticSessionControl.interpret_response(response). 
2.Because the flashing process will have a physical address and functional address distinction, so produced multiple clients, as you said, for the physical address and functional address coexistence, how to use a client to complete all the requests？
3.The current problem is that the canalyzer log is no problem, but in the udsoncan logging log, 8501 and 1001 have timeout problems，
how to enable the "isotp" logger in debug mode,can you share some code?

pylessard, 2023-09-20:
Ah ok, I did not noticed that you change the address as you go.
you can just change the address on the stack with `set_address()`. The client will keep using the same stack and everything will be wonderful.

to enable isotp log:``logging.getLogger('isotp').setLevel(logging.DEBUG)`

I don'T see the problem in the canalyzer log you shared.

pylessard, 2023-10-10:
Do you still need help with that issue?

hanlfa, 2023-10-10:
The problem is resolved, please close it

closed 2023-10-10 by pylessard