pylessard/python-udsoncan #172: Expect `suppress_positive_response` response NRC

zhongpipi, 2023-09-11:
The NRC response is expected when I request 'suppress_positive_response' massage. Like this:
```
        with pytest.raises(NegativeResponseException) as handle:
            with self.udsclient.suppress_positive_response:
                response = self.udsclient.start_routine(routine_id=routine_id, data=payload)
        assert handle.value.response.code == 0x33
```
But the code is 'return None', and will not check the response.
client.py 
```
        if request.suppress_positive_response or override_suppress_positive_response:
            return None
```

pylessard, 2023-09-18:
Yes, this is a limitation of the actual client, it works fully synchronously. We either return None or wait a certain amount of time in case a response happens. Maybe I could add a feature to do this wait.

Is it required in your case?

zhongpipi, 2023-09-18:
Thank you very much for your reply! 
According to ISO-14229, I think this situation exists. When we write test cases, we really need to test such situations. I would be grateful if you could add this feature!
Thank you very much for providing this open source library, which has been of great help to us.

pylessard, 2023-09-18:
Can you try this branch?
https://github.com/pylessard/python-udsoncan/tree/support-wait-nrc-when-suppress-pos-resp

```python
with client.suppress_positive_response(wait_nrc=True):
   pass
```

pylessard, 2023-09-18:
I'm waiting on your feedback to merge

zhongpipi, 2023-09-19:
Sorry for replying to you only now. Thank you very much for your modification. I tested it and it works very well and meets my needs very well. Thank you very very much!

pylessard, 2023-09-19:
Thank you for your feedback. Merged.

Fixed by #174

closed 2023-09-19 by pylessard

zhongpipi, 2023-09-19:
When will the official new version be released?

pylessard, 2023-09-19:
I have no plan. I can do it now if it helps you

zhongpipi, 2023-09-19:
Yes, I need the new version, downloaded via pip. grateful!

pylessard, 2023-09-19:
There you go sir. v1.19.0 is published on pypi.
Cheers

zhongpipi, 2023-09-19:
Thanks!
Have a great day or evening!

zhongpipi, 2023-09-19:
I use it like this:
`        with self.assertRaises(TimeoutError):
            with self.udsclient.suppress_positive_response(wait_nrc=True):
                self.udsclient.send_request(req)`

`                with pytest.raises(NegativeResponseException) as handle:
                    with uds.suppress_positive_response(wait_nrc=True):
                        uds.start_routine(routine_id=routine_id, data=data)
                assert handle.value.response.code == 0x13, 'nrc13 not raise'`