pylessard/python-udsoncan #177: Pass data to request_seed via unlock_security_access

pompushko, 2023-09-28:
Hello

Is it possible to pass some data to **request_seed**  via **unlock_security_access**?
As I see, **request_seed**  accept some data bytes.

Thank you!

pylessard, 2023-09-28:
Looks like I have forgotten to add that parameter on unlock_securityaccess.
Short term solution is to manually call request_seed / send_key  manually. 

I will add the missing parameter soon

pompushko, 2023-09-28:
> Looks like I have forgotten to add that parameter on unlock_securityaccess. Short term solution is to manually call request_seed / send_key manually.
> 
> I will add the missing parameter soon

Great! Thank you!

pylessard, 2023-10-01:
Fixed and publish into release v1.20.1

closed 2023-10-01 by pylessard

pompushko, 2023-10-10:
I have one question, is that right usage?

```
client.unlock_security_access(1, seed_params=bytes(0x1))
```

because when I use such way, in candump I see wrong data...
```
can0  18DA31F1   [4]  03 27 01 00
```

Thank you!

pylessard, 2023-10-10:
You just don't use the bytes object properly.  the bytes constructor, when given an integer n with create a bytes object that contains n zeros

you want to put your data into a list first : `bytes([0x00, 0x01])`