pylessard/python-udsoncan #180: Using J2534Connection: No data received

kirya-dev, 2023-10-05:
Code:
```python
import udsoncan
from udsoncan.connections import J2534Connection
from udsoncan.client import Client
from udsoncan.exceptions import *
from udsoncan.services import *

udsoncan.setup_logging()

conn = J2534Connection(
    windll="C:\programs\CHIPSOFT_J2534_Lite_Driver\j2534_interface.dll", 
    rxid=0x18DA_F1B0, 
    txid=0x18DA_B0F1,
    debug=True,
)

client_config={}

with Client(conn, request_timeout=100, config=client_config) as client:
   try:
        client.change_session(DiagnosticSessionControl.Session.extendedDiagnosticSession)
```
        

Logs:
```
2023-10-05 22:48:47 [INFO] Connection: J2534 FirmwareVersion: b'CHIPSOFT J2534 Lite v. 1.5.2, SN: 253FA666B98BEF79', dllVersoin: b'CHIPSOFT J2534 DLL Library v. 1.5.2, Jan  4 2021 12:19:06', apiVersionb'04.04'
2023-10-05 22:48:47 [INFO] Connection: J2534 Connection opened
2023-10-05 22:48:47 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Switching session to extendedDiagnosticSession (0x03)
2023-10-05 22:48:47 [DEBUG] Connection: Sending 2 bytes : [b'1003']
2023-10-05 22:48:48 [DEBUG] Connection: No data received: [TimeoutException] - Did not received response from J2534 RxQueue (timeout=1 sec)     
2023-10-05 22:48:48 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)   
2023-10-05 22:48:48 [INFO] Connection: J2534 Connection closed
```

pylessard, 2023-10-10:
Indeed, no data seems to have been received.
Do you have a log that shows what really happens on the bus?

kirya-dev, 2023-10-14:
It seems no support for ID29 bits

kirya-dev, 2023-10-15:
Im started PR
https://github.com/pylessard/python-udsoncan/pull/181

pylessard, 2023-10-15:
Thanks,
How do I know when this PR is ready?

kirya-dev, 2023-10-15:
I will report logs.

But right now new problems detected! Didnt support:
- Transmission of a Multi-Frame Request Message
- **Reception of a Multi-Frame Response Message**

Steps for reproduce:
```py
print(client.get_supported_dtc().service_data.dtcs)
```
in application - no response in Time.
and in logs:
```log
PassThruWriteMsgs(1, [{ISO15765; RxStatus: 0; TxFlags: ISO15765_FRAME_PAD | CAN_29BIT_ID; TS: 00:00:00; LEN: 6; ExtraIndex: 0; Data: 18-DA-B0-F1-19-0A}], 1=>1, 1000) -> STATUS_NOERROR
PassThruReadMsgs(1, [], 1=>0, 1) -> ERR_BUFFER_EMPTY
PassThruReadMsgs(1, [{ISO15765; RxStatus: TX_MSG_TYPE | TX_INDICATION | CAN_29BIT_ID; TxFlags: 0; TS: 00:00:00.0390000; LEN: 4; ExtraIndex: 0; Data: 18-DA-B0-F1}], 1=>1, 1) -> STATUS_NOERROR
PassThruReadMsgs(1, [{ISO15765; RxStatus: START_OF_MESSAGE | CAN_29BIT_ID; TxFlags: 0; TS: 00:00:00.0500000; LEN: 4; ExtraIndex: 0; Data: 18-DA-F1-B0}], 1=>1, 1) -> STATUS_NOERROR
PassThruReadMsgs(1, [], 1=>0, 1) -> ERR_BUFFER_EMPTY
PassThruReadMsgs(1, [], 1=>0, 1) -> ERR_BUFFER_EMPTY
PassThruReadMsgs(1, [], 1=>0, 1) -> ERR_BUFFER_EMPTY
...
```

No any reaction on START_OF_MESSAGE 😯

kirya-dev, 2023-10-15:
Works great with **Ioctl_ID.ISO15765_STMIN.value, 0**
Also in PR i add new method for formatting ID (ISO-15031-6)

Code example
```py
with Client(conn, request_timeout=100, config=client_config) as client:
    try:
        client.change_session(DiagnosticSessionControl.Session.defaultSession)
        dtc_data = client.get_supported_dtc().service_data
        for i in range(0, dtc_data.dtc_count):
            print(dtc_data.dtcs[i].id_iso())
```
            
```
PS C:\Users\33gee\Desktop\can-v2-udsocan> py app.py
2023-10-15 14:08:35 [INFO] Connection: J2534 FirmwareVersion: b'CHIPSOFT J2534 Lite v. 1.5.2, SN: 253FA666B98BEF79', dllVersoin: b'CHIPSOFT J2534 DLL Library v. 1.5.2, Jan  4 2021 12:19:06', apiVersionb'04.04'
2023-10-15 14:08:35 [INFO] Connection: J2534 PassThruConnect: b'Operation success'
2023-10-15 14:08:35 [INFO] Connection: J2534 PassThruIoctl SET_CONFIG: b'Operation success'
2023-10-15 14:08:35 [INFO] Connection: J2534 PassThruIoctl CLEAR_MSG_FILTERS: b'Operation success'
2023-10-15 14:08:35 [INFO] Connection: J2534 PassThruStartMsgFilter: b'Operation success'
2023-10-15 14:08:35 [INFO] Connection: J2534 PassThruIoctl CLEAR_RX_BUFFER: b'Operation success'
2023-10-15 14:08:35 [INFO] Connection: J2534 PassThruIoctl CLEAR_TX_BUFFER: b'Operation success'
2023-10-15 14:08:35 [INFO] Connection: J2534 Connection opened
2023-10-15 14:08:35 [DEBUG] Connection: Sending 2 bytes : [b'1001']
2023-10-15 14:08:35 [DEBUG] Connection: Received 2 bytes : [b'5001']
2023-10-15 14:08:35 [DEBUG] Connection: Sending 2 bytes : [b'190a']
2023-10-15 14:08:35 [DEBUG] Connection: Received 399 bytes : [b'.........']
U0415-68
U0416-9A
C1C00-F0
U0418-92
U0416-92
P2583-92
U1280-00
...
```

Nice to look😊

kirya-dev, 2023-10-15:
Now ready for merge PR or review.

pylessard, 2023-11-06:
merged

closed 2023-11-06 by pylessard