pylessard/python-udsoncan #182: How to implement the canfd flash based on the python-can library

hanlfa, 2023-10-17:
How to implement the canfd flash based on the python-can library

hanlfa, 2023-10-18:
my code:
bus = can.interface.Bus(interface="kvaser",channel=0, fd=True,   timing=can.BitTimingFd.from_sample_point(
    f_clock=80_000_000,
    nom_bitrate=500_000,
    nom_sample_point=75.0,
    data_bitrate=8_000_000,
    data_sample =  70 ))

config = {
            'p2_timeout':5, #0.05，1
            'request_timeout': 20, #5.1,25
            #'p2_star_timeout': 5,
            'use_server_timing': False, 
        }
        isotp_params = {
            'stmin':20,
            'blocksize': 8,  
            'tx_padding': 0xAA, 
            'rx_flowcontrol_timeout': 1000,  
            'rx_consecutive_frame_timeout': 1000,  
            'max_frame_size': 4095 
        }

        # txid = diag_funcreq- 7DF
        tp_addr_fun = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=diag_funcreq,rxid=diag_phyresp)  # Network
        stack = isotp.CanStack(bus=self.bus, address=tp_addr_fun,params=isotp_params)  # 
        conn = PythonIsoTpConnection(stack)  # 
        with Client(conn, config=config) as client:
              client.change_session(3)

pylessard, 2023-10-18:
there's a parameter in the isotp params that you can set called `can_fd`
See: https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#can_fd

pylessard, 2023-11-06:
Should we close that?

hanlfa, 2023-11-06:
ok,close it发自我的手机-------- 原始邮件 --------发件人： Pier-Yves Lessard ***@***.***>日期： 2023年11月6日周一 13:09收件人： pylessard/python-udsoncan ***@***.***>抄送： hanlfa ***@***.***>, Author ***@***.***>主    题： Re: [pylessard/python-udsoncan] How to implement the canfd flash based on the python-can library (Issue #182)
Should we close that?

closed 2023-11-06 by pylessard