pylessard/python-udsoncan #183: Logging configuration changes behavior of transfer buffers?

Upgreydd, 2023-10-18:
Hey,
While using `transfer_data()` I've faced `Transmit buffer full` exception. 
Due to that, I was going to debug it, so I've enabled logging through: 
```
udsoncan.setup_logging(path.join(path.dirname(__file__), "logging.conf"))
```
but it still happened so I've decided to make log more detailed. I've changed logging.conf
```
[logger_root]
level=ERROR
handlers=consoleHandler
```
to
```
[logger_root]
level=DEBUG
handlers=consoleHandler
```
and my code works. Every time I change `DEBUG` to `ERROR` my code fails - can it be related to some race condition in code?

Upgreydd, 2023-10-19:
@pylessard I'm sending a few MB of data in a loop and with `level=DEBUG` all is fine, not sure how it's related but looks like some backend handling of buffer cleanup fails. I've added `bus.flush_tx_buffer()` before each `client.transfer_data(seq, data)` but it doesn't fix the issue :/

pylessard, 2023-10-19:
Hmm I see, you dump a lot of data into the socketcan queue and by forcing a timeout value, you tell python-can to wait for socketcan before sending data.  You can leverage the rate limiter of the isotp TransportLayer instead. It will avoid dumping all the data into the python-can layer at once.

https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#rate-limiter

The rate limiter was introduced to solve a similar issue, but with a different hardware. See https://github.com/pylessard/python-can-isotp/issues/61

pylessard, 2023-10-19:
Also, if you want to control the send parameter, you do not have to modify python can. The TransportLayer let you specify your own send/recv function. You're probably using the default one in the CanStack object, but you don't have to. 

https://github.com/pylessard/python-can-isotp/blob/master/isotp/protocol.py#L1084-L1108

Instantiate the TransportLayer object directly and write your own rxfn/txfn.  You can base your work on the CanStack I wrote.

closed 2023-10-20 by Upgreydd