pylessard/python-udsoncan #186: ReadDataByIdentifier.py TimeoutException Did not receive response in time

ichbinbork, 2023-11-10:
Hi. I was trying to use read DID service. I have sent valid request  without any exception and from server side I can the message that coming from client side. Then I'm sending proper Positive Response Message to the server. After 5 second ('p2_timeout': 5 in configs.py) I received   "[TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=5.000 sec). Did not receive response in time. Global request timeout time has expired (timeout=5.000 sec)
PcanBus was not properly shut down" Error message couple of times. 

Here is the message that I sent from the client side : 03 22 00 01 00 00 00 00

Here is the positive response message that I sent from the server side:  06 62 01 00 00 00 00 00

I inspect the source code of your project more than one but I couldn't figure out what cause that server response can't reach the client.

In addition to them I want to inform you about defining data identifiers. I couldn't define did with this method
 "
config = dict(udsoncan.configs.default_client_config)
config['data_identifiers'] = {
   0x1234 : MyCustomCodecThatShiftBy4,    # Uses own custom defined codec. Giving the class is ok
   0x1235 : MyCustomCodecThatShiftBy4(),  # Same as 0x1234, giving an instance is good also
   0xF190 : udsoncan.AsciiCodec(15)       # Codec that read ASCII string. We must tell the length of the string
   }" 
Some how I couldn't change the 'data_identifiers' key value with this way. I manually updated the configs.py instead of it. 

Best regards

pylessard, 2023-11-12:
Hi,
For the timeout, the data have not reached the udsoncan module it's probably one of these:

- Data isn't received at all. a CAN log can show us
- The isotp layer does not receive the data. Either bad addresses are configured or another piece of code is depleting the CAN layer receive queue. You can enable debug logging for a logger named "isotp", you will get a good picture
- Timeouts value are too strict and the data arrives too late. In the 2020 version of the standard, the server is able to specify the timeouts value when setting the diagnostic session. This timing can be ignored by a parameter `use_server_timing=False`.

As for the config, sorry, but this makes no sense. You shouldn't edit the module code.
I don't know what you mean by "you can't", but I'm pretty sure that if you read the error you get, you will see why. All this piece of code do is assign an object to a dictionary entry, this is basic Python.

ichbinbork, 2023-11-13:
Hello again,

`use_server_timing=False` solved my problems about problems about response. 

Secondly, I agree with you. It was completely unreasonable. Then I checked my code couple of times and it fixed too. Thanks for help.

closed 2023-11-13 by ichbinbork