pylessard/python-udsoncan #187: send uds req with suppress postive response bit failed

xueshawu, 2023-11-14:
CAN Device : Vector 1630.
using the examples.rst code, 
send req: 10 03, it's ok, 
send req: 10 83, the CANOE  Trace view show failed. 
the python logger: DEBUG: Sendding 2 bytes :[b' 1083]，logger and CANoe Bus error  as below picture
[image]
[image]

pylessard, 2023-11-14:
Start by validating that you can simply sends raw CAN messages. Maybe you have the wrong bitrate or need to configure your bus object as CAN FD.  The issues is certainly below the UDS or the ISOTP layer.

xueshawu, 2023-11-16:
bitrate is configed ok, because  send 10 03 ok(client.change_seesion(3)), like blow code
with Clinet(conn) as client:
    client.change_session(3)

but when enable the suppress postive response bit,   send 10 83 failed. like blow code
``
with Clinet(conn) as client:
  with client.suppress_postive_response:
       client.change_session(3)

pylessard, 2023-11-17:
There's no link between suppress positive response and can level error. I can't help unless you do some serious diagnostic sorry

closed 2023-12-22 by pylessard