pylessard/python-udsoncan #189: How to set stmin when use transfer_data()

trandangkhoadev, 2023-11-29:
I'm using Transport layer and PythonIsoTpConnection but when I call transfer_data() the stmin will respect the stmin requirement of receiver. 
In case stmin from receiver as 0. The transfer_data() will send the cantp frame to fast such as stmin= 0.2ms. So how can I increase the delay between 2 frames in this case.

pylessard, 2023-11-29:
Can you explain why it is too fast?  
 - Is it because the receiver can not take it?  In that case, the receiver should not set stmin = 0
 - Is it because you overflow the transmit queue? In that case look at the rate limiter

You are quite lucky, because I happen to have added a parameter to override the receiver stmin tonight.  This is still under developement and will be part of the unreleased isotp v2.x.  The parameter will be `override_receiver_stmin`

https://github.com/pylessard/python-can-isotp/compare/v2.x...document_new_params#diff-1e3ed06e90fe3bf5ecf311ac1bd22215144cfc8c0924f5e05e3394573d630cebR32

pylessard, 2023-11-29:
Moving this issue to python-can-isotp module