pylessard/python-udsoncan #191: IsoTPSocketConnection can not be closed with tpsock in blocking mode

bacarz, 2023-12-04:
Closing IsoTPSocketConnection stucks on `self.rxthread.join()` when `self.tpsock` is created in blocking mode:
```
    tpsock = isotp.socket(timeout=0)
    connector = IsoTPSocketConnection(interface=interface, rxid=endpoint.tx, txid=endpoint.rx, tpsock=tpsock)
```
`rxthread_task` use directly socket.recv by `self.tpsock.recv()` . Is it possible to use selector?
```
    def rxthread_task(self):
        sel = selectors.DefaultSelector()
        sel.register(self.tpsock, selectors.EVENT_READ, data=None)
        while not self.exit_requested:
            events = sel.select(timeout=1)
            for _, _ in events:
                try:
                    data = self.tpsock.recv()
                    if data is not None:
                        self.rxqueue.put(data)
                except socket.timeout:
                    pass
                except Exception:
                    self.exit_requested = True
```
Or it would be better to make some changes to `isotp.socket.recv` directly?

pylessard, 2023-12-04:
Good point. It can certainly be improved to use a selector. I am actually reworking that section of code, will keep that in mind. thanks

pylessard, 2024-01-28:
Want to try this branch?
https://github.com/pylessard/python-udsoncan/tree/use-selector-in-socket-connection

bacarz, 2024-01-30:
Tested. Looks good.

closed 2024-01-30 by pylessard

nada-ben-ali, 2024-05-02:
@pylessard After adding the selectors in IsoTPSocketConnection, I think it won't work when using custom sockets that don't have a file descriptor, so events will always be None, and so self.rxqueue will always remain empty, which is the case for me.

pylessard, 2024-05-02:
Why would a socket have no file descriptor?
Are you able to provide a reproducing code?