pylessard/python-udsoncan #200: Performance tests

pompushko, 2024-01-22:
Hello

I use IsoTPSocketConnection for own project. My CanBis speed is 500kb\s

When I trying to make some tests with one request from ECU, I have such speed:
`100%|###########################################| 100 of 100 Time:  0:00:00 107.7 it/s`

[image]

[image]

With PythonIsoTpConnection too slow :D

[image]

But with cangen I can load almost fully my CAN-bus:
[image]

Main question, is that possible to make faster transmission? As I know, Python is limited by 1 code only. But I have 4. I tried to compile my test app with Cython and that didn't helped me :)

Thank you

pylessard, 2024-01-23:
Download latest isotp package (v2.0.3)
Update udsoncan to 1.21.1    (1.21.2 will release soon)

You should get an improvement

pylessard, 2024-01-23:
Just so you know, when you'll update udsoncan, make sure to pass an ``isotp.Address`` if you use the ``IsotpSocketConnection``. IT is now required. See https://udsoncan.readthedocs.io/en/latest/udsoncan/questions_answers.html#my-isotpsocketconnection-raises-an-error-after-updating-udsoncan

pompushko, 2024-01-23:
> Just so you know, when you'll update udsoncan, make sure to pass an `isotp.Address` if you use the `IsotpSocketConnection`. IT is now required. See https://udsoncan.readthedocs.io/en/latest/udsoncan/questions_answers.html#my-isotpsocketconnection-raises-an-error-after-updating-udsoncan

Well, This is strange, I have this code:
`    conn = IsoTPSocketConnection(args.interface, isotp.Address(isotp.AddressingMode.NormalFixed_29bits, rxid=0xF1, txid=args.ecuaddress), tpsock=tpsock)`

And I still get:
`RuntimeError: Provide an isotp.Address to the IsoTPSocketConnection. The interface has changed in a non-backward compatible way and this is now required.`

pylessard, 2024-01-23:
Are you sure this line raise the issue. check your stack trace.
[image]

pompushko, 2024-01-23:
Well.... Yes 
95 line...

[image]

[image]

pylessard, 2024-01-24:
I don't know what to say. Something else is going on. The test condition is quite straight forward, I even have unit test that validates it.

print kwargs inside the IsotpSocketConenction constructor please.

pompushko, 2024-01-24:
> I don't know what to say. Something else is going on. The test condition is quite straight forward, I even have unit test that validates it.
> 
> print kwargs inside the IsotpSocketConenction constructor please.

Could you, please,  tell me how to do that?
Thank you

pylessard, 2024-01-24:
edit the file defining `IsotpSocketConnection `.  You can find it by looking at ``udsoncan.connections.__file__``

then do ``print(kwargs)`` inside of  ``__init__``

pompushko, 2024-01-24:
Hehe...

```
{'rxid': 32, 'txid': 241}
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "udstest/uds.py", line 95, in udstest.uds.main
  File "/home/flasher/.local/lib/python3.11/site-packages/udsoncan/connections.py", line 306, in __init__
    raise RuntimeError(
RuntimeError: Provide an isotp.Address to the IsoTPSocketConnection. The interface has changed in a non-backward compatible way and this is now required.
```

pompushko, 2024-01-24:
Well. I found the problem. 

But, my tests give me same performance :(

pylessard, 2024-01-24:
What do you pass to PythonIsoTPConnection?   Try a notifierBasedCanStack.

pompushko, 2024-01-24:
I use IsotpSocketConnection , not PythonIsoTPConnection

pylessard, 2024-01-24:
You opened an issue saying that the PythonIsotpConnection was slower than IsoTPSocketConenction.

What are you asking?

pompushko, 2024-01-24:
Im asking, is that possible to use all CPU cores when using IsoTPSocketConnection or PythonIsoTpConnection. Or make faster transmission...

pylessard, 2024-01-24:
Using more CPU cores will not give you a faster transmission. The bottleneck is not the CPU.

To have a faster transmission, start by identifying the cause of the slowness. IsoTP protocol has some timing parameter that can be adjusted. Most likely that the receiving party is requiring the sender to not go too fast.   

Start by sharing a can log of a transmission and indicate what you consider slow.

pylessard, 2024-01-30:
Closing this issue. reopen if needed

closed 2024-01-30 by pylessard