pylessard/python-udsoncan #206: When using CAN as a diagnostic channel, when other application messages are sent periodically on the bus, calling the client.change_session interface will block the self.conn.empty_rxqueue() interface.

jjundot, 2024-02-03:
When using CAN as a diagnostic channel, when other application messages are sent periodically on the bus, calling the client.change_session interface will block the self.conn.empty_rxqueue() interface.

The specific location is shown in the figure
[image]

Since I also need to read some application packets for other purposes at the same time, is there any other better way without filtering out the application packets?

pylessard, 2024-02-03:
What are the package versions? What Connection  do you use?
Can you enable debug logging and share the logs?
If you use the ``isotp`` module, enable debug logging for it as well.

pylessard, 2024-02-03:
I have good hypothesis for that issue. 

It's possibly a mix of 2 issues, on that will be solved by this [PR ](https://github.com/pylessard/python-can-isotp/pull/116) and also that you are providing a rxfn that blocks forever without respecting the timeout argument being passed to you. You need to fix the later, otherwise you will always end up with a dangling thread and the app won't exit.

jjundot, 2024-02-03:
Thank you so much.

I use the PythonIsoTpConnection (can-isotp 2.0.3)
The PR #116  above should be very effective,  in my environment, it is blocked here:
`empty_rxqueue`  ---> `self.isotp_layer.stop_receiving()` --->  `time.sleep(0.05)`

[image]

closed 2024-02-03 by jjundot