pylessard/python-udsoncan #207: cannot import name 'Literal' from 'typing' (/usr/lib/python3.7/typing.py)

philippoo66, 2024-02-11:
starting my 'app' I get

```
phil@raspberrypi:~/open3e2 $ py Open3Eclient.py -c vcan0 -r 256 -v
Traceback (most recent call last):
File "Open3Eclient.py", line 22, in
import Open3Eclass
File "/home/phil/open3e2/Open3Eclass.py", line 2, in
import udsoncan
File "/home/phil/.local/lib/python3.7/site-packages/udsoncan/init.py", line 11, in
from udsoncan.common.CommunicationType import *
File "/home/phil/.local/lib/python3.7/site-packages/udsoncan/common/CommunicationType.py", line 6, in
from udsoncan import tools
File "/home/phil/.local/lib/python3.7/site-packages/udsoncan/tools.py", line 4, in
from udsoncan.typing import IOConfigEntry, IOConfig
File "/home/phil/.local/lib/python3.7/site-packages/udsoncan/typing.py", line 2, in
from typing import Dict, Optional, Any, Callable, Union, Type, Literal
ImportError: cannot import name 'Literal' from 'typing' (/usr/lib/python3.7/typing.py)
```

udsoncan is 1.22.0, Python is 3.7.3, Raspbian 10 (buster)

anybody any idea? please help! @pylessard ?!

thank you very much!!
Phil

pylessard, 2024-02-11:
Let me see, Literal is a 3.8 construct. I might have broken backward compatibility by mistake

pylessard, 2024-02-11:
It is, it's a compatibility issue with python 3.7

pylessard, 2024-02-11:
Fixed in released 1.22.1
https://github.com/pylessard/python-udsoncan/releases/tag/v1.22.1

Thanks for the report.
Let me know if that works for you

philippoo66, 2024-02-11:
perfect!! working fine again.

thank you very much for the quick reaction and the immediate fix!! 

greetings!
Phil

closed 2024-02-11 by pylessard

pylessard, 2024-02-12:
FYI, I am enabling automated regression testing with Jenkins. 
I broke compatibility with non-3.10 versions few time sin the last month.

Cheers

philippoo66, 2024-02-12:
thx! do you know why on my Raspi nothing higher than Python 3.7 gets installed? is this related to Buster? or do I have to explicitly/manually 'force' a higer version to get installed?

pylessard, 2024-02-12:
Probably that the repository of precompiled binaries for buster only includes 3.7
You can compile the wanted version, it is not that hard.  

See how I build different version for CI
https://github.com/pylessard/python-udsoncan/blob/enable-jenkins-ci/Dockerfile