pylessard/python-udsoncan #211: Act as server and parse python-can messages

Upgreydd, 2024-02-14:
Hey,
I'm trying to act as a server-side with udsoncan but I'm facing some issues related to parsing.
I'm receiving can message with data:
```
0x716 [8] 03 22 d1 00 00 00 00 00
```
arbitration_id [DLC] DATA, where 03 - is msg length, 22 - service, d1 00 is content and 00 00 00 00 is padding.
When I try to use: `Request.from_payload()` it cannot parse because the service is on the second position, not the first. I can of course use data[1:] to pass this failure, but then I'm trying to do a response like:
```
resp = Response(req.service, code=Response.Code.PositiveResponse, data=bytes.fromhex("01")
can_msg = can.Message(arbitration_id=0x71E, dlc=8, data=resp.get_payload())
bus.send(can_msg)
```
What I'm sending is:
```
0x71E [8] 62 01
```
What I'm expecting to send is:
```
0x71E [8] 03 62 d1 00 01
```
I'm not very familiar with can, but the lacking parts in the payload are length (03) and service (d1 00)
What I did do wrong?

pylessard, 2024-02-14:
Hi
You are passing an isotp payload directly to the UDS layer. You are lacking a layer of interpretation. The first byte is meant to be consumed by that isotp (transport layer). You can look at my other project for an implementation of that layer. If you input the whole Can message on one side, you will received the UDS payload on the other 

Let me know if I can help, you have some doc reading to do first I believe :)

Upgreydd, 2024-02-14:
@pylessard thank you - but which one project you're talking about? :D

Upgreydd, 2024-02-14:
@pylessard sorry for asking stupid questions, but as I understand I need to implement custom `rxfn` and `txfn` functions in isotp transport layer?

Upgreydd, 2024-02-14:
@pylessard Thank you mate :* Probably have all what I need. Have a nice evening.

closed 2024-02-14 by Upgreydd