pylessard/python-udsoncan #215: WriteDataByIdentifier service execution returned a negative response IncorrectMessageLengthOrInvalidFormat (0x13) when data length is correct

funnyrabbitvu, 2024-02-27:
Hi @pylessard,

While using service 2E, I received NRC 13 even the data length is correct and used bytes format.
[NegativeResponseException] : WriteDataByIdentifier service execution returned a negative response IncorrectMessageLengthOrInvalidFormat (0x13)
Server refused our request for service WriteDataByIdentifier with code "IncorrectMessageLengthOrInvalidFormat" (0x13)

Please help me to check this. Thank you!

pylessard, 2024-02-27:
I cannot check anything, you haven't shared anything.
Share code and log to get help

funnyrabbitvu, 2024-02-27:
> I cannot check anything, you haven't shared anything. Share code and log to get help
> 
> The negative response comes from your ECU, not this lib

Hi, I tried writing 16 bytes data to the ECU's DID (defined by the ECU). I also defined this DID codec in my configurations.

client.write_data_by_identifier(0xC001, b"\x00\x20\x00 \x00\x00\x25\x00\x000\x00\x00%\x00\x00\x00")

funnyrabbitvu, 2024-02-27:
> I cannot check anything, you haven't shared anything. Share code and log to get help
> 
> The negative response comes from your ECU, not this lib

Hi, I tried writing 16 bytes data to the ECU's DID (defined by the ECU). I also defined this DID codec in my configurations.

client.write_data_by_identifier(DID, b"\x00\x20\x00 \x00\x00\x25\x00\x000\x00\x00%\x00\x00\x00")

pylessard, 2024-02-27:
Your ECU refuses it. I can't do much.
Check your log

funnyrabbitvu, 2024-02-29:
Checked issue from ECU side. I will close this issue. Thank you!

closed 2024-02-29 by funnyrabbitvu