pylessard/python-udsoncan #216: Adding parameter to Communication Control Service(0x28)

ichbinbork, 2024-02-29:
Hi,

First of all its not a issue, this is a question about modifying udsoncan library. 

As I see you haven't add the "nodeIdentificationNumber" parameter for Communication Control service. I need this parameter for my project. Before forking and diving into library I wanted suggestions from you. Ofc I can do it in a manual way but I want to save some time. 

Best regards.

pylessard, 2024-03-01:
I reviewed the doc. This parameter has been introduced in the 2013 version of the standard. My initial version of this lib was based on the 2006 version, I never noticed this was added and nobody asked for it. 
I will implement during the weekend, I will have time

pylessard, 2024-03-01:
I have started the work here : https://github.com/pylessard/python-udsoncan/tree/add_node_id_number_to_comm_control

It's untested

ichbinbork, 2024-03-01:
Yes, I can imagine that this is a kind of niche parameter to be used. The main reason why I need this parameter is that I have such a prototype architecture 
[image]
Suppose I have three CAN subnets under one ECU as shown in the picture. I need this parameter to control the Rx Tx of each CAN line. Also this parameter needs to be two bits because some ECUs have 3 of them.

It is off-topic right now but I am thinking of adding a new session in the coming days. According to 14229-1 
[image]
I can define private sessions between 0x40 and 0x3F. Can you also give instructions (best practices) on this?

Sincerely regards

pylessard, 2024-03-01:
I see. You will have enough bits, the parameter is 16 bits

I don't think there is best practices for adding a session. Or I'm not aware of any. Just pick a number

ichbinbork, 2024-03-01:
As I said adding a session is not relative to this topic. I want to prepare the base for future projects. That why I want to create separate OEM specific sessions.

pylessard, 2024-03-02:
Implementation is complete.
Wanna try this branch : add_node_id_number_to_comm_control ?

ichbinbork, 2024-03-04:
It seems okay here is the log:
[image]

I appreciate this modification but I also want to create a custom session. Could you give directions about it? 

Regards

closed 2024-03-04 by ichbinbork

pylessard, 2024-03-04:
Hi, good news, thank you
For the session, there is nothing to do on the client side. All the work happens on the server side.  Just do

```
client.change_session(42)
```

cheers

pylessard, 2024-03-05:
Available in release 1.23.0