pylessard/python-udsoncan #228: SocketConnection cannot be closed

nada191, 2024-03-28:
Closing SocketConnection stucks on self.rxthread.join().

sock:
[image]
connector = SocketConnection(sock=sock)

It looks like [this issue](https://github.com/pylessard/python-udsoncan/issues/191) but for SocketConnection and not IsoTPSocketConnection (for IsoTPSocketConnection there is no stuck) using release 1.22.1 and reproducible with 1.23.0 as well.

By putting the self.sock.settimeout(0.1) back into the SocketConnection constructor, the problem was solved. It was removed when the selector was added in this [MR](https://github.com/pylessard/python-udsoncan/pull/203/files).

pylessard, 2024-04-01:
There is a unit test that checks that this work.
Can you show me how you trigger that issue?

pylessard, 2024-04-01:
Is it possible that you call open and close one after the other without waiting?
I think there's a weakness in that case

Try to reproduce your error and then try this branch : fix_socket_connection_that_blocks_on_open_close

let me know the outcome

nada-ben-ali, 2024-04-03:
Problem trigger:

* sock = socketPool.get(ip, port, connection_name) # socketPool is a pool of open tcp sockets in the test, so it checks if the socket is still open otherwise it creates it if it's closed or if it's a new socket.

* Create the connector using the returned socket: SocketConnection(sock=sock)

* create the client by opening the connection using socketConnection.open

* send a diag after checking that the connection is not closed using SocketConnection.specific_send(payload, timeout=None) --> no problem with sending the diag

* we've received a positive response

* it stucks when closing the connection in self.rxthread.join().

I've tested your branch and the same problem still occurs. When opening the connection, self.started_event.is_set() was not True, so it closes the connection and then stucks in self.rxthread.join().

            
 [image]

pylessard, 2024-04-03:
Something does not make sense, You say that the the thread never starts but you are able to send and receive.

Are you able to write a demo script that recreate the issue and share that with me?
I just don't see how this can happen...

nada-ben-ali, 2024-04-03:
The thread never starts using the branch you provided with which I was unable to send or receive, it just gets stuck when opening the connection. But without the branch, I'm able to send and receive, but when I close the connection, it stucks.

I will try to make a small  script.

pylessard, 2024-05-27:
Following up on this, update?
I still never reproduced your issue. Would need some kind of code that reproduce.

pylessard, 2024-07-02:
If no update, I will close this issue

closed 2024-07-09 by pylessard