pylessard/python-udsoncan #229: P2 timer issue on client.transfer_data(block_number, data

stanislavZemek, 2024-04-04:


stanislavZemek, 2024-04-04:
Hello,
at first, very nice library which works with socketcan. 
I probably found a problem with this library. 
I am sending 2048 bytes of data to my developed ECU via UDS protocol from tester program which contains your python library.
After receiving first block of data from PEAK cable my ECU sent confirm frame back to tester in properly time, but the udsoncan library either wrote warning: 
**2024-04-04 13:31:03 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=0.200 sec)**

Sniffed communication:
 (1712230262.871388)  can0       760   [8]  18 02 36 01 00 00 01 20
 (1712230262.873197)  can0       768   [3]  30 00 00
 (1712230262.873296)  can0       760   [8]  21 A1 06 01 08 45 06 01
 (1712230262.873443)  can0       760   [8]  22 08 4B 06 01 08 51 06
 (1712230262.873588)  can0       760   [8]  23 01 08 57 06 01 08 5D
 (1712230262.873726)  can0       760   [8]  24 06 01 08 00 00 00 00
 (1712230262.873864)  can0       760   [8]  25 00 00 00 00 00 00 00
 (1712230262.874002)  can0       760   [8]  26 00 00 00 00 00 E1 40
 (1712230262.874158)  can0       760   [8]  27 01 08 63 06 01 08 00
 (1712230262.874296)  can0       760   [8]  28 00 00 00 C1 42 01 08
****** next messages ********
 (1712230262.934464)  can0       760   [8]  2E 60 00 BF 14 37 BD 46
 (1712230262.934738)  can0       760   [8]  2F 80 BC 70 47 00 BF 00
 (1712230262.934872)  can0       760   [8]  20 ED 00 E0 80 B4 00 AF
 (1712230262.935191)  can0       760   [8]  21 04 4B DB 68 1B 0A 03
 (1712230262.935418)  can0       760   [8]  22 F0 07 03 18 46 BD 46
 (1712230262.935647)  can0       760   [8]  23 80 BC 70 47 00 BF 00
 (1712230262.935874)  can0       760   [8]  24 ED 00 E0 80 B4 83 B0
 (1712230263.103335)  can0       768   [3]  02 76 01

As you can see last row contain response from my unit and the response time from last message is is lower than P2 time. 
Do you have any idea where should be a problem? 
Can be that a problem with buffering inside PEAK cable? 
Thank you very much for your response in advance and wish you good luck with projects like this.

pylessard, 2024-04-04:
Regarding the initial message, The timeout of 0.2sec is a bit of a tight fit. Enable python logging for Debug. The time you show in your can log shows times from start to end of the transmission of the transport layer, maybe the application layer has a latency to read the underlying layer. Since you don't have much margin, not impossible that it is the case. logs will tell us.

For the 2nd question, see this excerpt from ISO-14229
[image]

stanislavZemek, 2024-04-04:
Thank you very much for your fast reply :+1: 
I increased the P2 timer for 500ms and it works properly now, but it's strange that it did not accept "response pending" packet when I tried with P2 timer 100ms before attempt with 200ms.
I am absolutely beginner with Python. I commonly used C for microcontrollers so I don't know if I am able to properly logging data. :smile: Will try.

Thank you very much for explanation of P2 resolution.. I missed that.. I will correct that in my uds library in C on the server.

pylessard, 2024-04-04:
To enable logging, you can do something like

```
import logging
logging.basicConfig(level=logging.DEBUG)
```

pylessard, 2024-04-04:
I forgot!
To have time printed out, you can do this.

```
logging.basicConfig(level=logging.DEBUG, format='%(asctime)s [%(levelname)s] <%(name)s> %(message)s')
```

stanislavZemek, 2024-04-05:
Hello,

thank you so much for your help. I will switch back the timer to 100ms and will send you a log.

stanislavZemek, 2024-04-08:
Hello,
I tried to get working the logging , but I think you wanted much advanced log. Correct? I think the output is the same as without logging. I did not find any .log file so I think Python is sending all information directly to command line.
python3 uds_flasher_STM32.py 
2024-04-08 14:06:50 [INFO] Connection: Connection opened
2024-04-08 14:06:50 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Switching session to extendedDiagnosticSession (0x03)
2024-04-08 14:06:50 [DEBUG] Connection: Sending 2 bytes : [1003]
2024-04-08 14:06:50 [DEBUG] Connection: Received 6 bytes : [500300640064]
2024-04-08 14:06:50 [INFO] UdsClient: Received positive response for service DiagnosticSessionControl (0x10) from server.
2024-04-08 14:06:50 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Received new timing parameters. P2=0.100s and P2*=1.000s.  Using these value from now on.
2024-04-08 14:06:50 [INFO] UdsClient: RoutineControl<0x31> - ControlType=0x01 - Starting routine ID 0xff00 (EraseMemory) with a payload of 9 bytes
2024-04-08 14:06:50 [DEBUG] Connection: Sending 13 bytes : [3101ff00440801000000004a8c]
2024-04-08 14:06:50 [DEBUG] Connection: Received 3 bytes : [7f3178]
2024-04-08 14:06:50 [DEBUG] Connection: Received 5 bytes : [7101ff0001]
2024-04-08 14:06:50 [INFO] UdsClient: Received positive response for service RoutineControl (0x31) from server.
2024-04-08 14:06:50 [INFO] UdsClient: RequestDownload<0x34> - Requesting a download (client to server) for memory location [Address=0x8010000 (32 bits), Size=0x6000 (32 bits)] and DataFormatIdentifier 0x00 (Compression:0x0, Encryption:0x0)
2024-04-08 14:06:50 [DEBUG] Connection: Sending 11 bytes : [3400440801000000006000]
2024-04-08 14:06:50 [DEBUG] Connection: Received 4 bytes : [74200026]
2024-04-08 14:06:50 [INFO] UdsClient: Received positive response for service RequestDownload (0x34) from server.
2024-04-08 14:06:50 [INFO] UdsClient: TransferData<0x36> - Sending a block of data with SequenceNumber=1 that is 2048 bytes long .
2024-04-08 14:06:50 [DEBUG] Connection: Sending 2050 bytes : [360100000120010a01087d09010883090108890901088f0901089509010800000000000000000000000000000000d15101089b09010800000000b1530108f9530108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108a9090108bd090108490a0108490a0108490a0108490a0108d1090108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a010800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005ff8e0f100000000000000000000000010b5054c237833b9044b13b10448aff300800123237010bd6c00002000000000785a010808b5034b1bb103490348aff3008008bd0000000070000020785a010880b584b000af03463a60fb710b46bb80174800f09cfff860fb68002b1dd0fb68032b1ad8fb79134a1361124b00229a60104b0022da60bb880e4a13600d4b00221a750d4b3a680b49094800f0b1fe0346002b08d000f0ecf905e0fb68032b02d900f04af8ffe700bf1037bd4680bd00bfa400002088000020a000002080b58cb000af0346fb710b46bb8013467b800123bb62fb79fb610023bb61bb885b01bb600023fb607b885b013b614ff6ff737b6100233b6201237b621b23fb6207f108031946034800f01efd00bf3037bd4680bda400002080b500af4ff6f0724ff4ec610120fff7cbff4ff6ff7240f2df710220fff7c4ff00bf80bd80b500af0a4800f0defc00f0a5f8fff7e5ff074800f0c3fd0221054800f04fff0346002b01d000f087f900bf80bd00bfa400002080b500af62b600bf00f072fb00f04ef800f0e8f800f0bcf800f084f8fff7c4ff194800f0a2fd4ff47a7302221749184802f006fc4ff47a7000f08cfb4ff47a7302221249124802f0fbfb4ff47a7000f081fb0e4a4ff4e0610220fff73bff02f06ffd0c4a00210c4802f0d0fd03460b4a13600b4a00210b4802f0c8fd03460a4a136002f08ffdfee7a400002000000020cc000020d85a0108d505010814010020fc5a0108090601081801002080b590b000af07f1180328220021184605f048fa3b1d00221a605a609a60da601a610223bb610123bb621023fb6200237b6307f11803184601f0acfc0346002b01d000f009f90f237b600023bb600023fb6000233b6100237b613b1d0021184601f01aff0346002b01d000f0f5f800bf4037bd4680bd000080b500af174b184a1a60164b02225a60144b00229a60134b0022da60114b4ff4a0221a610f4b00225a610e4b00221a760c4b01225a760b4b01229a76094b0022da76084b00221a77064b01225a77054800f012fb0346002b01d000f0c1f800bf80bd00bfa40000200064004080b500af114b124a1a60104b4ff4e1325a600e4b00229a600c4b0022da600b4b00221a61094b0c225a61084b00229a61064b0022da61054802f0eafa0346002b01d000f097f800bf80bd00bfcc0000200044004080b588b000af07f1100300221a605a609a60da601d4b9b691c4a43f0400393611a4b9b6903f04003fb60fb68174b9b69164a43f004039361144b9b6903f00403bb60bb68114b9b69104a43f0080393610e4b9b6903f008037b607b68002240210b4801f0e3fb40233b6101237b610023bb610223fb6107f110031946044801f075f900bf2037bd4680bd00bf001002400018014080b584b000af7860094a07f1080392e8030083e8030007f108031a464ff4e0610820fff71bfe4ff47a7002f05dfdebe7ac5a010880b586b000af7860094a07f10c0307ca03c31a8007f10c031a4640f201710220fff702fe40f6b83002f044fdece700bfb45a010880b582b000af78607b681b68044a934201d100f013fa00bf0837bd4680bd00bf002c014080b400af72b600bffee7000080b584b000af184b9b69174a43f001039361154b9b6903f00103bb60bb68124bdb69114a43f08053d3610f4bdb6903f080537b607b6800220f216ff0010001f0c6f80a4b5b68f [... cut ...]
  File "/home/standa/.local/lib/python3.10/site-packages/udsoncan/client.py", line 976, in transfer_data
    response = self.send_request(request)
  File "/home/standa/.local/lib/python3.10/site-packages/udsoncan/client.py", line 2213, in send_request
    raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' %
udsoncan.exceptions.TimeoutException: Did not receive response in time. P2 timeout time has expired (timeout=0.100 sec)

Here is udsoncan.log
[udsoncan.log](https://github.com/pylessard/python-udsoncan/files/14905204/udsoncan.log)

pylessard, 2024-04-08:
Hi, thank you for this.
Unfortunately, I forgot to give you the parameter to include millisecond timing in the log. Sorry.

Can you setup your logging like this? (I forgot ``%(msecs)03d`` ) 

```
logging.basicConfig(level=logging.DEBUG, format='%(asctime)s.%(msecs)03d [%(levelname)s] <%(name)s> %(message)s')

```

stanislavZemek, 2024-04-08:
It looks like it's necessary to edit your code as well, because the answer is from your library and does not contain milliseconds. Only output from my own code contains milliseconds. Where should I add msecs to your code?
Thank you.

stanislavZemek, 2024-04-08:
Okay, I will doublecheck that. For now first two rows are in milliseconds, but it's not usefull for us. 

2024-04-08 15:11:56.106 INFO:	Logging Info
2024-04-08 15:11:56.106 DEBUG:	Logging Debug
2024-04-08 15:11:56 [INFO] Connection: Connection opened
2024-04-08 15:11:56 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Switching session to extendedDiagnosticSession (0x03)
2024-04-08 15:11:56 [DEBUG] Connection: Sending 2 bytes : [1003]
2024-04-08 15:11:56 [DEBUG] Connection: Received 6 bytes : [500300640064]
2024-04-08 15:11:56 [INFO] UdsClient: Received positive response for service DiagnosticSessionControl (0x10) from server.
2024-04-08 15:11:56 [INFO] UdsClient: DiagnosticSessionControl<0x10> - Received new timing parameters. P2=0.100s and P2*=1.000s.  Using these value from now on.
2024-04-08 15:11:56 [INFO] UdsClient: RoutineControl<0x31> - ControlType=0x01 - Starting routine ID 0xff00 (EraseMemory) with a payload of 9 bytes
2024-04-08 15:11:56 [DEBUG] Connection: Sending 13 bytes : [3101ff00440801000000004a8c]
2024-04-08 15:11:56 [DEBUG] Connection: Received 3 bytes : [7f3178]
2024-04-08 15:11:56 [DEBUG] Connection: Received 5 bytes : [7101ff0001]
2024-04-08 15:11:56 [INFO] UdsClient: Received positive response for service RoutineControl (0x31) from server.
2024-04-08 15:11:56 [INFO] UdsClient: RequestDownload<0x34> - Requesting a download (client to server) for memory location [Address=0x8010000 (32 bits), Size=0x6000 (32 bits)] and DataFormatIdentifier 0x00 (Compression:0x0, Encryption:0x0)
2024-04-08 15:11:56 [DEBUG] Connection: Sending 11 bytes : [3400440801000000006000]
2024-04-08 15:11:56 [DEBUG] Connection: Received 4 bytes : [74200026]
2024-04-08 15:11:56 [INFO] UdsClient: Received positive response for service RequestDownload (0x34) from server.
2024-04-08 15:11:56 [INFO] UdsClient: TransferData<0x36> - Sending a block of data with SequenceNumber=1 that is 2048 bytes long .
2024-04-08 15:11:56 [DEBUG] Connection: Sending 2050 bytes : [360100000120010a01087d09010883090108890901088f0901089509010800000000000000000000000000000000d15101089b09010800000000b1530108f9530108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108a9090108bd090108490a0108490a0108490a0108490a0108d1090108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a010800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005ff8e0f100000000000000000000000010b5054c237833b9044b13b10448aff300800123237010bd6c00002000000000785a010808b5034b1bb103490348aff3008008bd0000000070000020785a010880b584b000af03463a60fb710b46bb80174800f09cfff860fb68002b1dd0fb68032b1ad8fb79134a1361124b00229a60104b0022da60bb880e4a13600d4b00221a750d4b3a680b49094800f0b1fe0346002b08d000f0ecf905e0fb68032b02d900f04af8ffe700bf1037bd4680bd00bfa400002088000020a000002080b58cb000af0346fb710b46bb8013467b800123bb62fb79fb610023bb61bb885b01bb600023fb607b885b013b614ff6ff737b6100233b6201237b621b23fb6207f108031946034800f01efd00bf3037bd4680bda400002080b500af4ff6f0724ff4ec610120fff7cbff4ff6ff7240f2df710220fff7c4ff00bf80bd80b500af0a4800f0defc00f0a5f8fff7e5ff074800f0c3fd0221054800f04fff0346002b01d000f087f900bf80bd00bfa400002080b500af62b600bf00f072fb00f04ef800f0e8f800f0bcf800f084f8fff7c4ff194800f0a2fd4ff47a7302221749184802f006fc4ff47a7000f08cfb4ff47a7302221249124802f0fbfb4ff47a7000f081fb0e4a4ff4e0610220fff73bff02f06ffd0c4a00210c4802f0d0fd03460b4a13600b4a00210b4802f0c8fd03460a4a136002f08ffdfee7a400002000000020cc000020d85a0108d505010814010020fc5a0108090601081801002080b590b000af07f1180328220021184605f048fa3b1d00221a605a609a60da601a610223bb610123bb621023fb6200237b6307f11803184601f0acfc0346002b01d000f009f90f237b600023bb600023fb6000233b6100237b613b1d0021184601f01aff0346002b01d000f0f5f800bf4037bd4680bd000080b500af174b184a1a60164b02225a60144b00229a60134b0022da60114b4ff4a0221a610f4b00225a610e4b00221a760c4b01225a760b4b01229a76094b0022da76084b00221a77064b01225a77054800f012fb0346002b01d000f0c1f800bf80bd00bfa40000200064004080b500af114b124a1a60104b4ff4e1325a600e4b00229a600c4b0022da600b4b00221a61094b0c225a61084b00229a61064b0022da61054802f0eafa0346002b01d000f097f800bf80bd00bfcc0000200044004080b588b000af07f1100300221a605a609a60da601d4b9b691c4a43f0400393611a4b9b6903f04003fb60fb68174b9b69164a43f004039361144b9b6903f00403bb60bb68114b9b69104a43f0080393610e4b9b6903f008037b607b68002240210b4801f0e3fb40233b6101237b610023bb610223fb6107f110031946044801f075f900bf2037bd4680bd00bf001002400018014080b584b000af7860094a07f1080392e8030083e8030007f108031a464ff4e0610820fff71bfe4ff47a7002f05dfdebe7ac5a010880b586b000af7860094a07f10c0307ca03c31a8007f10c031a4640f201710220fff702fe40f6b83002f044fdece700bfb45a010880b582b000af78607b681b68044a934201d100f013fa00bf0837bd4680bd00bf002c014080b400af72b600bffee7000080b584b000af184b9b69174a43f001039361154b9b6903f00103bb60bb68124bdb69114a43f08053d3610f4bdb6903f080537b607b6800220f216ff0010001f0c6f80a4b5b68fb60fb6823f0e063fb60fb6843f08073fb60044afb68536000bf1037bd4680bd00bf001002400000014 [... cut ...]
    return func(self, *args, **kwargs)
  File "/home/standa/.local/lib/python3.10/site-packages/udsoncan/client.py", line 976, in transfer_data
    response = self.send_request(request)
  File "/home/standa/.local/lib/python3.10/site-packages/udsoncan/client.py", line 2213, in send_request
    raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' %
udsoncan.exceptions.TimeoutException: Did not receive response in time. P2 timeout time has expired (timeout=0.100 sec)

stanislavZemek, 2024-04-08:
I had already set logging with udsoncan.setup_logging() :smile: :smile: 
So I've had overwrite previous logging settings.

Here is the correct log:
2024-04-08 15:17:52.395 INFO:	Logging Info
2024-04-08 15:17:52.395 DEBUG:	Logging Debug
2024-04-08 15:17:52.396 INFO:	Connection opened
2024-04-08 15:17:52.396 INFO:	DiagnosticSessionControl<0x10> - Switching session to extendedDiagnosticSession (0x03)
2024-04-08 15:17:52.396 DEBUG:	Sending request to server
2024-04-08 15:17:52.396 DEBUG:	Sending 2 bytes : [1003]
2024-04-08 15:17:52.396 DEBUG:	Waiting for server response
2024-04-08 15:17:52.398 DEBUG:	Received 6 bytes : [500300640064]
2024-04-08 15:17:52.398 DEBUG:	Received response from server
2024-04-08 15:17:52.398 INFO:	Received positive response for service DiagnosticSessionControl (0x10) from server.
2024-04-08 15:17:52.398 INFO:	DiagnosticSessionControl<0x10> - Received new timing parameters. P2=0.100s and P2*=1.000s.  Using these value from now on.
2024-04-08 15:17:52.399 INFO:	RoutineControl<0x31> - ControlType=0x01 - Starting routine ID 0xff00 (EraseMemory) with a payload of 9 bytes
2024-04-08 15:17:52.399 DEBUG:		Payload data : 440801000000004a8c
2024-04-08 15:17:52.399 DEBUG:	Sending request to server
2024-04-08 15:17:52.399 DEBUG:	Sending 13 bytes : [3101ff00440801000000004a8c]
2024-04-08 15:17:52.399 DEBUG:	Waiting for server response
2024-04-08 15:17:52.470 DEBUG:	Received 3 bytes : [7f3178]
2024-04-08 15:17:52.470 DEBUG:	Received response from server
2024-04-08 15:17:52.470 DEBUG:	Server requested to wait with response code RequestCorrectlyReceived_ResponsePending (0x78), single request timeout is now set to P2* (1.000 seconds)
2024-04-08 15:17:52.470 DEBUG:	Waiting for server response
2024-04-08 15:17:52.678 DEBUG:	Received 5 bytes : [7101ff0001]
2024-04-08 15:17:52.678 DEBUG:	Received response from server
2024-04-08 15:17:52.679 INFO:	Received positive response for service RoutineControl (0x31) from server.
2024-04-08 15:17:52.679 INFO:	RequestDownload<0x34> - Requesting a download (client to server) for memory location [Address=0x8010000 (32 bits), Size=0x6000 (32 bits)] and DataFormatIdentifier 0x00 (Compression:0x0, Encryption:0x0)
2024-04-08 15:17:52.679 DEBUG:	Sending request to server
2024-04-08 15:17:52.679 DEBUG:	Sending 11 bytes : [3400440801000000006000]
2024-04-08 15:17:52.679 DEBUG:	Waiting for server response
2024-04-08 15:17:52.683 DEBUG:	Received 4 bytes : [74200026]
2024-04-08 15:17:52.684 DEBUG:	Received response from server
2024-04-08 15:17:52.684 INFO:	Received positive response for service RequestDownload (0x34) from server.
2024-04-08 15:17:52.685 INFO:	TransferData<0x36> - Sending a block of data with SequenceNumber=1 that is 2048 bytes long .
2024-04-08 15:17:52.685 DEBUG:	Data to transfer : 00000120010a01087d09010883090108890901088f0901089509010800000000000000000000000000000000d15101089b09010800000000b1530108f9530108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108a9090108bd090108490a0108490a0108490a0108490a0108d1090108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a0108490a010800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005ff8e0f100000000000000000000000010b5054c237833b9044b13b10448aff300800123237010bd6c00002000000000785a010808b5034b1bb103490348aff3008008bd0000000070000020785a010880b584b000af03463a60fb710b46bb80174800f09cfff860fb68002b1dd0fb68032b1ad8fb79134a1361124b00229a60104b0022da60bb880e4a13600d4b00221a750d4b3a680b49094800f0b1fe0346002b08d000f0ecf905e0fb68032b02d900f04af8ffe700bf1037bd4680bd00bfa400002088000020a000002080b58cb000af0346fb710b46bb8013467b800123bb62fb79fb610023bb61bb885b01bb600023fb607b885b013b614ff6ff737b6100233b6201237b621b23fb6207f108031946034800f01efd00bf3037bd4680bda400002080b500af4ff6f0724ff4ec610120fff7cbff4ff6ff7240f2df710220fff7c4ff00bf80bd80b500af0a4800f0defc00f0a5f8fff7e5ff074800f0c3fd0221054800f04fff0346002b01d000f087f900bf80bd00bfa400002080b500af62b600bf00f072fb00f04ef800f0e8f800f0bcf800f084f8fff7c4ff194800f0a2fd4ff47a7302221749184802f006fc4ff47a7000f08cfb4ff47a7302221249124802f0fbfb4ff47a7000f081fb0e4a4ff4e0610220fff73bff02f06ffd0c4a00210c4802f0d0fd03460b4a13600b4a00210b4802f0c8fd03460a4a136002f08ffdfee7a400002000000020cc000020d85a0108d505010814010020fc5a0108090601081801002080b590b000af07f1180328220021184605f048fa3b1d00221a605a609a60da601a610223bb610123bb621023fb6200237b6307f11803184601f0acfc0346002b01d000f009f90f237b600023bb600023fb6000233b6100237b613b1d0021184601f01aff0346002b01d000f0f5f800bf4037bd4680bd000080b500af174b184a1a60164b02225a60144b00229a60134b0022da60114b4ff4a0221a610f4b00225a610e4b00221a760c4b01225a760b4b01229a76094b0022da76084b00221a77064b01225a77054800f012fb0346002b01d000f0c1f800bf80bd00bfa40000200064004080b500af114b124a1a60104b4ff4e1325a600e4b00229a600c4b0022da600b4b00221a61094b0c225a61084b00229a61064b0022da61054802f0eafa0346002b01d000f097f800bf80bd00bfcc0000200044004080b588b000af07f1100300221a605a609a6 [... cut ...]
 (1712582272.748833)  can0  760   [8]  2E 4B DB 69 0A 4A 23 F0
 (1712582272.749017)  can0  760   [8]  2F 00 73 D3 61 4F F4 40
 (1712582272.749262)  can0  760   [8]  20 71 08 48 01 F0 F8 F9
 (1712582272.749489)  can0  760   [8]  21 13 20 01 F0 52 F8 14
 (1712582272.749847)  can0  760   [8]  22 20 01 F0 4F F8 00 BF
 (1712582272.749980)  can0  760   [8]  23 08 37 BD 46 80 BD 00
 (1712582272.750206)  can0  760   [8]  24 64 00 40 00 10 02 40
 (1712582272.829463)  can0  768   [4]  03 7F 36 78
 (1712582272.923409)  can0  768   [3]  02 76 01

stanislavZemek, 2024-04-08:
I've changed "latency" parameter inside my program and the result is the same:
Could be a problem that the uds client library send last CAN message to TX buffer to PEAK cable and then immediately starts count time to 100ms without "sent check"? 
Thank you :-) 

last few rows of sniffed communication:
 (1712584566.234136)  can0  760   [8]  22 20 01 F0 4F F8 00 BF
 (1712584566.234364)  can0  760   [8]  23 08 37 BD 46 80 BD 00
 (1712584566.234592)  can0  760   [8]  24 64 00 40 00 10 02 40
 (1712584566.293713)  can0  768   [4]  03 7F 36 78
 (1712584566.408078)  can0  768   [3]  02 76 01

stanislavZemek, 2024-04-08:
By the way? Do you know what is common P2 Timer in UDS-CAN devices?  I saw 150ms in devices which I have.
The communication is working well from around 500ms.

pylessard, 2024-04-08:
Please reshare both full logs with a timeout that is long enough

For the correct value, it's tricky. it's a matter of the interval boundary location. 
ISO-14229 says end of transmit, begin of receive. But this repo does begin of transmit, end of receive.  It is possible to have end of transmit/end of receive by using a blocking send.

stanislavZemek, 2024-04-08:
Here are logs of communication between my bootloader of ECU and UDS client. 
I bet it's caused by filling of buffer as you mentioned that you start timeout when you begin transmit. I'll check if there is possible to set TX buffer only for one message as parameter of PEAK.
Logs are in attachment, because they are large. 
Thank you for your support. :+1: 
[python_log.txt](https://github.com/pylessard/python-udsoncan/files/14907535/python_log.txt)
[can_sniff.txt](https://github.com/pylessard/python-udsoncan/files/14907536/can_sniff.txt)

stanislavZemek, 2024-04-09:
Hello Pylessard,

thank you. I appreciate your help!
The response is long, because timer P2 doesn't require sending of "response pending frame" during writing to flash memory. I will set P2 timer to 500ms and it will be sufficient. 
I am using this type of connection - conn = IsoTPSocketConnection('can0', isotp.Address(isotp.AddressingMode.Normal_11bits, rxid=0x768, txid=0x760))
How can I add the parameter to IsoTPSocketConnection setup? To be honest I don't know if I am using your isotp implementation or isotp socket. :sweat_smile: According to connection setup from your example I bet I am using isotp socket. 
Or should I set that during can0 initialization?

pylessard, 2024-04-09:
yeah ok, you would find allt he information in the doc. I can recap quickly.
The comm stack is Application -> Transport -> DataLink   (uds -> isotp -> can).   
On linux, you can do isotp/can with a kernel module that you interact with with a socket. By using ``IsoTPSocketConnection``, you use that.
You can also use a pure python implementation of isotp that I wrote, it will send the data to python-can, another library to do CAN in python.

You can make your own socket and give it to the ``IsoTPSocketConnection`` object.  You want to create a socket with the ``WAIT_TX_DONE `` flag set and give it to the connection. Something likethis (untested)

``` python
import isotp
from udsoncan.connections import IsoTPSocketConnection
sock = isotp.socket()
flags = socket.get_opts().optflag
sock.set_opts(optflags=socket.get_opts().optflag | isotp.flags.WAIT_TX_DONE)
conn = IsoTPSocketConnection('can0', address, tpsock=sock)
```

But looking at the doc, this flags blocks the send() method until the data is passed down to the CAN layer. The CAN layer will most likely buffers the data, emptying the isotp layer. Since you have a blocksize of 0,  all the data will be sent a once. This might not give much improvement actually...   that is a design weakness

ISO-15765 specify that each lyer should inform the upper layer when it is finished, but neither my python implementation nor the kernel module does it.

If you think the issue is resolved, please close it.
Thanks

stanislavZemek, 2024-04-09:
Hello Pylessard,

thank you for your final summary. I will try to create my own socket. If I will be successful I paste here a solution.
Now it's absolutely clear and thank you very much for your help Your python program is cool.

Standa.

closed 2024-04-09 by stanislavZemek

funnyrabbitvu, 2024-08-29:
Hi @stanislavZemek @pylessard I want to ask about memory location to check memory after finished transfering driver data. I used method client.start_routine(RID, byte_checksum) after request transfer exit, I also calculated crc of all data chunk transfered and it matched with crc calculated from .s19 file, but it seemed ECU return incorrectResult then I was not able to clear Application memory (received NRC24). Was it because I was checking in wrong memloc? Do I need to point to that particular memloc?

To clear memory: client.start_routine(0xFF00, byte 0x44 + byte memory address + byte memory size)

Thank you guys.

stanislavZemek, 2024-08-29:
Hello @funnyrabbitvu,
if I understand correctly, you are attempting to erase memory inside some ECU via UDS protocol. Due to mentioned NRC I think you are not in programming UDS session. I don't know if your ECU require Secure Access (0x27) to start erase memory routine. The start address of the program and the memory size depends on the ECU manufacturer.

pylessard, 2024-08-29:
For different problems, please open a new issue. Also you need to shares can logs + description of the issue you encountered.

Note that all ECU are differents, and a given NRC can happens in different conditions based on the ecu.