pylessard/python-udsoncan #230: Runtime Error: Connection is not open

p31d4, 2024-04-13:
Hi,

if I run the following code:

`
tpsock = isotp.socket(timeout=0.1)
tpsock.bind(...
conn = IsoTPSocketConnection(..., tpsock = tpsock)
client = Client(conn,...
client.tester_present()
`

The message is successfully sent, but I receive the error:
RuntimeError: Connection is not open.

Besides that, I would also have a problem calling the method "close", because self.rxthread.join() would be called in a thread which was never started.

I fixed that locally changing the method "open", from the class IsoTPSocketConnection in the file .../udsoncan/connections.py

`
self.tpsock = tpsock
if self.tpsock.bound:
    self.rxthread = threading.Thread(target=self.rxthread_task, daemon=True)
    self.rxthread.start()
    self.opened = True
` 

Based on that, I would like to ask:
- am I missing something, and maybe the problem is already fixed somehow?
- why the "address" parameter is not Optional in the IsoTPSocketConnection constructor? In this case for example it's is useless, it was already provided in the tpsock.bind call.

Also as a side note, I now you are probably very busy, but the documentation for version 1.23.0 is outdated, in the examples the IsoTPSocketConnection constructor is still using rxid, txid.

Thanks and regards.

pylessard, 2024-04-27:
The way it's design, you pass the address in the constructor, then open() calls bind. Do that and it should solve all your problems

p31d4, 2024-05-22:
@pylessard thanks for the replay, actually I tried that, in my first approach I was following the examples (ipsis literi, using the "with ... as client" block) and the problem described was always there. I came up with the code I described here after going deep in the implementation. My goal reporting it was mainly to  offer a solution in case anyone else had the same problem, because with the fix I did locally I could finish my stuff. Thx anyway.

closed 2024-05-22 by p31d4