pylessard/python-udsoncan #231: TimeoutError: timed out

p31d4, 2024-04-13:
Hi,

I'm trying to send a can message in a virtualized Linux machine (VM on native Windows), and a keep receiving a timeout in tpsock/__init__.py line 108 (self._socket.send(data,flags)) when I run the following code:

```
with Client(conn,  request_timeout=2, config=config) as client:
    client.tester_present()
```
The bus is working properly (I've tested it with can-utils) and the problem is somehow related to the "select" API, on method "rxthread_task". I've also tried the branch "fix_socket_connection_that_blocks_on_open_close", but the problem was still there.

I fixed the problem locally re-writiing the method "rxthread_task" as:

```
while True:
    try:
        time.slee(0.2)
        data = self.tpsock.recv()
        if data is None:
            break
        else:
            self.rxqueue.put(data)
    except Exception:
        pass
```

Looks like the "select" API is blocking forever, but I'm not entirely sure.

Does someone already experienced this timeout issue and has any clue how to fix it without having to re-write the code?

Thanks and regards

pylessard, 2024-04-27:
Possible that you don't open the connection properly like in the other issue?
If you want help to debug, share how you configure your setup.

p31d4, 2024-05-22:
@pylessard thanks for the replay, from my side this issue can be close.

closed 2024-05-27 by pylessard