pylessard/python-udsoncan #232: Problem with waiting time to response longer than 1,658 seconds

stanislavZemek, 2024-04-15:
Hello Pylessard,

sorry to bother you again.
I don't know what I am doing wrong, but I have another issue with timing of messages. :facepalm: :smile: 
When I have sent a routine request to erase memory on my server, my server had started erasing. During memory erasing, the server sent NRC with code 0x78. Firstly it is sent within P2max and another one is sent within P2*time. Despite sending NRC 0x78, the server only accepts this message a few times or not at all (it depends on the length of P2*time) and the udsoncan is stopping accept NRC 0x78 after 1.658 seconds which have elapsed from the request.
Below you can find two examples of communication.
Thank you for your time.

**Example of P2*time == 1.0s:**
DEBUG:
2024-04-15 19:58:14.155 INFO:	RoutineControl<0x31> - ControlType=0x01 - Starting routine ID 0xff00 (EraseMemory) with a payload of 9 bytes
2024-04-15 19:58:14.155 DEBUG:		Payload data : 440801000000004a8c
2024-04-15 19:58:14.155 DEBUG:	Sending request to server
2024-04-15 19:58:14.155 DEBUG:	Sending 13 bytes : [3101ff00440801000000004a8c]
2024-04-15 19:58:14.155 DEBUG:	Waiting for server response
2024-04-15 19:58:14.496 DEBUG:	Received 3 bytes : [7f3178]
2024-04-15 19:58:14.497 DEBUG:	Received response from server
2024-04-15 19:58:14.497 DEBUG:	Server requested to wait with response code RequestCorrectlyReceived_ResponsePending (0x78), single request timeout is now set to P2* (1.000 seconds)
2024-04-15 19:58:14.497 DEBUG:	Waiting for server response
2024-04-15 19:58:15.192 DEBUG:	Received 3 bytes : [7f3178]
2024-04-15 19:58:15.193 DEBUG:	Received response from server
2024-04-15 19:58:15.193 DEBUG:	Waiting for server response
2024-04-15 19:58:15.888 DEBUG:	Received 3 bytes : [7f3178]
2024-04-15 19:58:15.888 DEBUG:	Received response from server
2024-04-15 19:58:15.889 DEBUG:	Waiting for server response
2024-04-15 19:58:16.155 DEBUG:	No data received: [TimeoutException] - Did not received ISOTP frame in time (timeout=0.2664806842803955 sec) 
2024-04-15 19:58:16.156 ERROR:	[TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=0.266 sec)
2024-04-15 19:58:16.303 INFO:	Connection closed

PHYSICAL COMMUNICATION:
 (1713203894.155669)  can0       760   [8]  10 0D 31 01 FF 00 44 08
 (1713203894.157507)  can0       768   [3]  30 00 00
 (1713203894.157680)  can0       760   [8]  21 01 00 00 00 00 4A 8C
 (1713203894.496325)  can0       768   [4]  03 7F 31 78
 (1713203895.192271)  can0       768   [4]  03 7F 31 78
 (1713203895.888202)  can0       768   [4]  03 7F 31 78
 (1713203896.584012)  can0       768   [4]  03 7F 31 78
 (1713203897.257912)  can0       768   [4]  03 7F 31 78
 (1713203897.954027)  can0       768   [4]  03 7F 31 78
 (1713203898.161608)  can0       768   [6]  05 71 01 FF 00 01

**Example of P2*time == 3.0s:**
DEBUG:
2024-04-15 20:01:51.166 INFO:	RoutineControl<0x31> - ControlType=0x01 - Starting routine ID 0xff00 (EraseMemory) with a payload of 9 bytes
2024-04-15 20:01:51.166 DEBUG:		Payload data : 440801000000004a8c
2024-04-15 20:01:51.166 DEBUG:	Sending request to server
2024-04-15 20:01:51.166 DEBUG:	Sending 13 bytes : [3101ff00440801000000004a8c]
2024-04-15 20:01:51.166 DEBUG:	Waiting for server response
2024-04-15 20:01:51.507 DEBUG:	Received 3 bytes : [7f3178]
2024-04-15 20:01:51.507 DEBUG:	Received response from server
2024-04-15 20:01:51.507 DEBUG:	Server requested to wait with response code RequestCorrectlyReceived_ResponsePending (0x78), single request timeout is now set to P2* (3.000 seconds)
2024-04-15 20:01:51.507 DEBUG:	Waiting for server response
2024-04-15 20:01:53.166 DEBUG:	No data received: [TimeoutException] - Did not received ISOTP frame in time (timeout=1.6584365367889404 sec) 
2024-04-15 20:01:53.166 ERROR:	[TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=1.658 sec)
2024-04-15 20:01:53.323 INFO:	Connection closed

PHYSICAL COMMUNICATION:
 (1713204111.166397)  can0       760   [8]  10 0D 31 01 FF 00 44 08
 (1713204111.168220)  can0       768   [3]  30 00 00
 (1713204111.168424)  can0       760   [8]  21 01 00 00 00 00 4A 8C
 (1713204111.506809)  can0       768   [4]  03 7F 31 78
 (1713204113.595039)  can0       768   [4]  03 7F 31 78
 (1713204115.172013)  can0       768   [6]  05 71 01 FF 00 01

pylessard, 2024-04-15:
There's a parameter for a global timeout that overrides all the other. It just make sure that the client does not block indefinetely, you can set it to ``None``.   There error message doe snot mention P2 nor P2*, but ``Global request timeout time has expired``

You either set it here : https://udsoncan.readthedocs.io/en/latest/udsoncan/client.html#request_timeout
Or you pass it to the client constructor.

stanislavZemek, 2024-04-16:
Hello Pylessard,

thank you for your help. I've overlooked parameter in client configuration... There was a parameter left from the example I used and I forgot about it. :facepalm: 
Now it works properly. :ok_hand: 
Have a nice day!

Standa.

closed 2024-04-16 by pylessard