pylessard/python-udsoncan #233: isotp error handler not propagated to uds connection

elupus, 2024-05-23:
When a isotp error occurs, it triggers a call to a callback, that one can supply to the constructor of the isotp stack called `error_handler`. The default of this callback is to just log an message to log.

The uds stack should be informed of this error and abort on-going requests in PythonIsoTpConnection.specific_wait_frame. With log request timeouts, it can take a long time for the request to abort. It also aborts with a TimeoutException instead of the actual low level fault.

elupus, 2024-05-23:
In general i might be that the protocol._stop_receiving() call is not propagated to the listener. not all error_handler calls abort the reception.

pylessard, 2024-05-26:
Thanks, I see what you mean. The error should be reported to the upper layer.
I will check this, but maybe not short term.