pylessard/python-udsoncan #235: TimeoutException and request timeout not respected?

pompushko, 2024-05-27:
@pylessard  Hello :)

I dont know, but sometimes my timeout settings wont respects...

```
                client = Client(conn, request_timeout=5)
                client.open()
                client.set_config('exception_on_negative_response', False)
                client.set_config('exception_on_timeout', False)
                client.set_config('p2_timeout', 10.0)
                client.set_config('p2_server_max', 10.0)
                client.set_config('request_timeout', 10.0)
                client.set_config('p2_star_timeout', 10.0)
                client.set_config('use_server_timing', False)
```

And sometimes my code throw this:
```[TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=1.968 sec)```

Why? Where is takes **timeout=1.968 sec**?
Thank you.

pylessard, 2024-05-27:
The error message is wrong. I can fix that.
You hit your 10s timeout.

What happens is that the timeout value is used to block the thread, if it unblock and there is still time to wait, a new timeout value is computed.  So 1.968 = 10-<time elapsed>. That mainly happens if your device sends a NRC 0x78, meaning it respond but ask to wait.

See here: https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/client.py#L2194

pompushko, 2024-05-27:
Hmm, so I have to extend global time to avoid long NRCs?

pylessard, 2024-05-27:
Yes, or you can disable the global timeout by setting it to None.  Just note that the device can block your client forever if you do so

pylessard, 2024-05-30:
Should we close this issue?

pompushko, 2024-05-31:
> Should we close this issue?

Well, if that is not an issue or fixed, yes. Please :D

pompushko, 2024-06-11:
> timeout=1.971 sec

well, but it think that its not 10 seconds :(

closed 2024-06-11 by pompushko

pylessard, 2024-06-11:
[image]

Look at the logs