pylessard/python-udsoncan #238: udsoncan.exceptions.TimeoutException: Did not receive response in time. P2 timeout time has expired

ep081106, 2024-06-15:
Hello pylessard:
  I want to use the ETAS 582.1 to read the DTC, but i got this error:
udsoncan.exceptions.TimeoutException: Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec);

this is my code:
[readTheDTCBy582.txt](https://github.com/user-attachments/files/15848009/readTheDTCBy582.txt)

then I got the following error:
"D:\Software\Program Files (x86)\Python39-32\python.exe" D:\Software\Python_Scripts\PythonProject\UdsOnCan_RE300\UDSdemo_582_noUDSOnCan_V0.1_20240614.py 
{'interface': 'etas', 'channel': 'ETAS://USB/ES582.1:115287/CAN:1'}
{'interface': 'etas', 'channel': 'ETAS://USB/ES582.1:115287/CAN:2'}
{'interface': 'etas', 'channel': 'ETAS://USB/VN1630:36922/CAN:2'}
{'interface': 'etas', 'channel': 'ETAS://USB/VN1630:36922/CAN:3'}
{'interface': 'etas', 'channel': 'ETAS://USB/VN1630:36922/CAN:4'}
{'interface': 'etas', 'channel': 'ETAS://VIRTUAL:VXL/CANCARD:0/CAN:1'}
{'interface': 'etas', 'channel': 'ETAS://VIRTUAL:VXL/CANCARD:0/CAN:2'}
CHANNEL_1:ETAS://USB/ES582.1:115287/CAN:1
CHANNEL_2:ETAS://USB/ES582.1:115287/CAN:2
bus:ETAS://USB/ES582.1:115287/CAN:1
Source path:... D:\Software\Python_Scripts\PythonProject\UdsOnCan_RE300\UDSdemo_582_noUDSOnCan_V0.1_20240614.py
16:34:10.038022 call        75 def read_dtcs_and_snapshots():
16:34:10.039526 line        76     with Client(conn, request_timeout=100, config=configRE300) as client:  # Application layer (UDS protocol)
    Source path:... D:\Software\Program Files (x86)\Python39-32\lib\site-packages\udsoncan\client.py
    Starting var:.. self = <udsoncan.client.Client object at 0x04279148>
    Starting var:.. conn = <udsoncan.connections.PythonIsoTpConnection object at 0x0425DAD8>
    Starting var:.. config = {'exception_on_negative_response': True, 'except...server_timing': True, 'extended_data_size': None}
    Starting var:.. request_timeout = 100
    16:34:10.039526 call       112     def __init__(self, conn: BaseConnection, config: ClientConfig = default_client_config, request_timeout: Optional[float] = None):
    16:34:10.040525 line       113         self.conn = conn
    16:34:10.040525 line       114         self.config = cast(ClientConfig, dict(config))  # Makes a copy of given configuration
    16:34:10.040525 line       117         if request_timeout is not None:
    16:34:10.040525 line       118             self.config['request_timeout'] = request_timeout
    16:34:10.040525 line       119         self.suppress_positive_response = Client.SuppressPositiveResponse()
    16:34:10.040525 line       120         self.payload_override = Client.PayloadOverrider()
    16:34:10.041525 line       121         self.last_response = None
    16:34:10.041525 line       123         self.session_timing = dict(p2_server_max=None, p2_star_server_max=None)
    16:34:10.041525 line       125         self.refresh_config()
    16:34:10.041525 return     125         self.refresh_config()
    Return value:.. None
    Starting var:.. self = <udsoncan.client.Client object at 0x04279148>
    16:34:10.041525 call       127     def __enter__(self):
    16:34:10.041525 line       128         self.open()
    16:34:10.044525 line       129         return self
    16:34:10.045263 return     129         return self
    Return value:.. <udsoncan.client.Client object at 0x04279148>
Source path:... D:\Software\Python_Scripts\PythonProject\UdsOnCan_RE300\UDSdemo_582_noUDSOnCan_V0.1_20240614.py
New var:....... client = <udsoncan.client.Client object at 0x04279148>
16:34:10.045263 line        77         response = client.get_dtc_by_status_mask(status_mask)
    Source path:... D:\Software\Program Files (x86)\Python39-32\lib\site-packages\udsoncan\client.py
    Starting var:.. self = <udsoncan.client.Client object at 0x04279148>
    Starting var:.. status_mask = 47
    16:34:10.045263 call      1243     def get_dtc_by_status_mask(self, status_mask: int) -> Optional[services.ReadDTCInformation.InterpretedResponse]:
    16:34:10.045690 line      1258         return self.read_dtc_information(services.ReadDTCInformation.Subfunction.reportDTCByStatusMask, status_mask=status_mask)
2024-06-15 16:34:10 [INFO] Connection: Connection opened

**2024-06-15 16:34:10 [INFO] UdsClient: ReadDTCInformation<0x19> - Sending request with subfunction "reportDTCByStatusMask" (0x02).
2024-06-15 16:34:54 [DEBUG] Connection: Sending 3 bytes : [19022f]**

**_"it took 44s to send the message"_**

2024-06-15 16:34:55 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive IsoTP frame from the Transport layer in time (timeout=1 sec) 
2024-06-15 16:34:55 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)
2024-06-15 16:34:55 [INFO] Connection: Connection closed
    16:34:55.656008 exception 1258         return self.read_dtc_information(services.ReadDTCInformation.Subfunction.reportDTCByStatusMask, status_mask=status_mask)
    Exception:..... udsoncan.exceptions.TimeoutException: Did not re.... P2 timeout time has expired (timeout=1.000 sec)
    Call ended by exception
Source path:... D:\Software\Python_Scripts\PythonProject\UdsOnCan_RE300\UDSdemo_582_noUDSOnCan_V0.1_20240614.py
16:34:55.657476 exception   77         response = client.get_dtc_by_status_mask(status_mask)
Exception:..... udsoncan.exceptions.TimeoutException: Did not re.... P2 timeout time has expired (timeout=1.000 sec)
16:34:55.657476 line       133             log_file.write('---------------------------End to read the Snapshot---------------------------' + '\n\n')
    Source path:... D:\Software\Program Files (x86)\Python39-32\lib\site-packages\udsoncan\client.py
    Starting var:.. self = <udsoncan.client.Client object at 0x04279148>
    Starting var:.. type = <class 'udsoncan.exceptions.TimeoutException'>
    Starting var:.. value = TimeoutException('Did not receive response in ti...P2 timeout time has expired (timeout=1.000 sec)')
    Starting var:.. traceback = <traceback object at 0x0426B608>
    16:34:55.657476 call       131     def __exit__(self, type, value, traceback):
    16:34:55.657476 line       132         self.close()
    16:34:55.669341 return     132         self.close()
    Return value:.. None
Source path:... D:\Software\Python_Scripts\PythonProject\UdsOnCan_RE300\UDSdemo_582_noUDSOnCan_V0.1_20240614.py
Call ended by exception
Elapsed time: 00:00:45.631319
Traceback (most recent call last):
  File "D:\Software\Python_Scripts\PythonProject\UdsOnCan_RE300\UDSdemo_582_noUDSOnCan_V0.1_20240614.py", line 136, in <module>
    read_dtcs_and_snapshots()
  File "D:\Software\Program Files (x86)\Python39-32\lib\site-packages\pysnooper\tracer.py", line 305, in simple_wrapper
    return function(*args, **kwargs)
  File "D:\Software\Python_Scripts\PythonProject\UdsOnCan_RE300\UDSdemo_582_noUDSOnCan_V0.1_20240614.py", line 77, in read_dtcs_and_snapshots
    response = client.get_dtc_by_status_mask(status_mask)
  File "D:\Software\Program Files (x86)\Python39-32\lib\site-packages\udsoncan\client.py", line 1258, in get_dtc_by_status_mask
    return self.read_dtc_information(services.ReadDTCInformation.Subfunction.reportDTCByStatusMask, status_mask=status_mask)
  File "D:\Software\Program Files (x86)\Python39-32\lib\site-packages\udsoncan\client.py", line 176, in decorated
    return func(self, *args, **kwargs)
  File "D:\Software\Program Files (x86)\Python39-32\lib\site-packages\udsoncan\client.py", line 1726, in read_dtc_information
    response = self.send_request(request)
  File "D:\Software\Program Files (x86)\Python39-32\lib\site-packages\udsoncan\client.py", line 2187, in send_request
    raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' %
udsoncan.exceptions.TimeoutException: Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)

[image]

could you please give me some help to fix it?  thanks very much!

pylessard, 2024-06-15:
It means that the device did not respond and there is a timeout while waiting for the response. We can see in the can log that there is indeed no response, a message with the second byte set to 99 should appear

ep081106, 2024-06-15:
> It means that the device did not respond and there is a timeout while waiting for the response. We can see in the can log that there is indeed no response, a message with the second byte set to 99 should appear

but the device works fine.  I can get the DTC by using cantool/busmaster and this device.

pylessard, 2024-06-15:
Can show me a log that works and a log that does not?
Did you mean that the library should wait 44 sec before triggering a timeout, if so you can adjust those values. Refers to the documentation

ep081106, 2024-06-17:
Hello pylessard，
please find the log in the following picture.
I want to read the DTC by using ETAS 582(CAN FD). 
[image]  

[image]

pylessard, 2024-06-17:
2 things
 1. I don't see the response in the "good" log. But you also have hidden the next message with the red square.
 2. PAdding byte is different. Does that make a difference for your device?  CC vs 00

ep081106, 2024-06-17:
1. 
[image]
2. my colleague told me，CC is for FD CAN.  00 is for CAN.

pylessard, 2024-06-17:
Ok, 
About padding, it really depends on what your device expects. 
You can change the padding byte with a parameter. Try that.

pylessard, 2024-06-17:
Also, I noticed that the bad log uses a different status mask. The device should respond anyway, but I wanted to point it out.

2F vs AF

ep081106, 2024-06-17:
> Ok, About padding, it really depends on what your device expects. You can change the padding byte with a parameter. Try that.

hello pylessard，i changed the padding and got the wrong log.

ep081106, 2024-06-17:
Hello pylessard.  
1. [image]
it takes a longe time to send the 1902AF.

2. and it didn't receive the response.
[image]

ep081106, 2024-06-17:
maybe i found this issues. i set the fd=True. but it sent by CAN. 
[image]
 if i send the $19 by can.message. i can get the response
[image]

closed 2024-06-17 by ep081106

pylessard, 2024-06-17:
Thank you for the update. Indeed this parameter might be useful to you.
https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#can_fd

Cheers

ep081106, 2024-06-17:
> Thank you for the update. Indeed this parameter might be useful to you. https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#can_fd
> 
> Cheers

Thanks for your help ! 
I added "can_fd" in the isotp_params and it works fine.

isotp_params = {
    'stmin': 32,  # 将请求发送方在连续帧之间等待时间。0 ~ 127ms或100 ~ 900ns，取值范围为0xf1 ~ 0xf9
    # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
    'blocksize': 8,  # 流控帧单包大小,0为不限制
    # Request the sender to send 8 consecutives frames before sending a new flow control message
    'wftmax': 0,  # 错误帧最大数
    # Number of wait frame allowed before triggering an error
    'tx_padding': 0xCC,  # 空白字节CAN是0x00, FD CANi用0xCC填充
    # Will pad all transmitted CAN messages with byte 0x00. None means no padding
    'rx_flowcontrol_timeout': 1000,  # 流控帧超时时间
    # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
    'rx_consecutive_frame_timeout': 1000,  # 连续帧超时时间
    # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
    'can_fd': True
}

xuechengz0801, 2024-06-18:
> but the device works fine. I can get the DTC by using cantool/busmaster and this device.

Hi,ep081106
Could you use 582.1 with filters on Pthyon-can to recieve the response only from ID 0x7CA?