pylessard/python-udsoncan #240: Can I use multiple RX addresses ?

ichbinbork, 2024-06-30:
Hey, 

I want to use multiple rx ids for responding to upcoming frames.

`tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x123, rxid=0x456)`

In the above block, the module only filters the 0x456 ID according to gives the response. But I want to filter more than one ID like can module in python. For instance, I want to filter "0x111", "0x222", "0x333" at the same time. Is it possible?

For changing address I created that function but I don't want to call it every process. I need different approach for handling multiple rxids.

```
    def set_can_addresses(self, physical_can_request_id, physical_can_response_id):
        try:
            txid = int.from_bytes(physical_can_request_id[:-1], 'big')
            rxid = int.from_bytes(physical_can_response_id[:-1], 'big')
            self.tp_addr = isotp.Address(isotp.AddressingMode.Normal_29bits, txid=txid, rxid=rxid)
            self.stack = isotp.NotifierBasedCanStack(bus=self.bus, notifier=self.notifier, address=self.tp_addr, params=isotp_params.isotp_params)
            self.stack.set_sleep_timing(0, 0)
        except Exception as e:
            print(e)
```

Best regards

pylessard, 2024-06-30:
What do you mean exactly? 
You want to send a request on a given ID then accept and answer from any of the ID in a list? Do you have prior knowledge of which ID will respond? If so, I suggest to simply create 3 instance of the CanStack. You can use 3 UDS clients or create a connection that will put the response of any CanStack into the same queue, then implement a recv function that reads from that Queue. It's not that hard. 

Also, you can change the address of a TransportLayer (e.g. the CanStack) by calling `set_address` instead of creating a new instance each time

ichbinbork, 2024-06-30:
"You want to send a request on a given ID then accept and answer from any of the ID in a list":  Yes exactly. Also I want to send them request. 

"Do you have prior knowledge of which ID will respond?". Yes, I have the response of each ECU ID to answer my request.

I will prefer the creating queue with upcoming responses. Cuz crating 3 different connection is little bit hard to handle (number may increase that's why hard to handle). Also I suspicious that the more than one client crash the stack according to issues of this repo.

Thanks for the advice. Is there a method for change baudrate like `set_address `?

pylessard, 2024-06-30:
Well having 3 full stack (client/connection/TransportLayer/Bus) is the simplest way. You shouldn't encounter any issue doing so. If I were you, I'd make a function that create and configure all these objects and return a working client. Then put those client in a dictionary index by an enum so I can do `clients[ECU1].some_request(some_param1, some_param2)`.   If you find it complex to create 3 clients, you will find it more complex to design a custom connection.

If you want to stick with the custom connection option, I suggest you look at how the PythonIsoTPConnectionV2 is designed an do something similar, bu write to a queue instead internally. 

To change the baudrate, I am not sure if you are talking about the UDS Baudrate control service or the underlying can bus bitrate. `udsoncan` supports the baudrate control service. To change the CAN layer, check tyhe python-can documentation.

pylessard, 2024-06-30:
yeah. Something like that could do.
You don't have to use `with` absolutely. Might be cleaner like this:

```python

client1.open()
client2.open()
client3.open()

client_1.change_session(1)
client_1.some_request(some_param1, some_param2)

client_2.change_session(1)
client_2.some_request(some_param1, some_param2)

client_3.change_session(1)
client_3.some_request(some_param1, some_param2)

client3.close()
client2.close()
client1.close()
```

Cheers

ichbinbork, 2024-06-30:
Thanks for your attention.

Best regards.

ichbinbork, 2024-07-01:
I have tried to call `set_address` method but I failed. Here is my block 
```
address = isotp.Address(isotp.AddressingMode.Normal_29bits,txid=0x18DDE910 , rxid=0x18DDF3E8)

   isotp.TransportLayerLogic.set_address(address=address)
```

I know that I need to create instance of TransportLayerLogic class first but when I try to create it according to exception I need to set rxfn and txfn. 

Also I want to change my address after connection created is it possible ?

ichbinbork, 2024-07-01:
Thanks again

closed 2024-07-01 by ichbinbork