pylessard/python-udsoncan #244: Did not received ISOTP frame

mpersisth, 2024-09-13:
[root@localhost test]# python3 example.py
2024-09-13 16:23:04 [INFO] Connection: Connection opened
2024-09-13 16:23:04 [INFO] UdsClient: ECUReset<0x11> - Requesting reset of type 0x01 (hardReset)
2024-09-13 16:23:04 [DEBUG] Connection: Sending 2 bytes : [1101]
2024-09-13 16:23:05 [DEBUG] Connection: No data received: [TimeoutException] - Did not received ISOTP frame in time (timeout=1 sec)
2024-09-13 16:23:05 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)

C[root@localhost uds-server-master]# candump vcan1 -a
  vcan1  733   [3]  02 11 01                  '...'
  vcan1  73B   [8]  03 7F 11 11 00 00 00 00   '........'

I confirm that the response message was sent back within one second. Given this information, what potential issues could have caused this behavior?

pylessard, 2024-09-13:
Bad address configuration could be a cause.
You can share how you setup your UDS stack, it'll be easier to review.

What connection de you use?
Can you enable timestamp in you logs?

mpersisth, 2024-09-14:
Sorry, I'm not quite sure about the correct procedure to include timestamps in logs.                                                                                                                                                             
TJZ_config = {                                                                                                                                                     
'exception_on_negative_response' : False,                                                                                                                          
'exception_on_invalid_response' : False,                                                                                                                           
'exception_on_unexpected_response' : False,                                                                                                                        
'security_algo' : None,                                                                                                                                            
'security_algo_params' : None,                                                                                                                                     
'tolerate_zero_padding' : False,                                                                                                                                   
'ignore_all_zero_dtc' : True,                                                                                                                                      
'dtc_snapshot_did_size' : 2, # Not specified in standard. 2 bytes matches other services format.                                                                   
'server_address_format' : None, # 8,16,24,32,40                                                                                                                    
'server_memorysize_format' : None, # 8,16,24,32,40                                                                                                                 
'data_identifiers' : {0xf190:""},                                                                                                                                  
'input_output' : {},                                                                                                                                               
'request_timeout': 5,                                                                                                                                              
'p2_timeout': 0.5,                                                                                                                                                 
'p2_star_timeout': 5,                                                                                                                                              
'use_server_timing' : False                                                                                                                                        
}                                                                                                                                                                  
                                                                                                                                                                   
                                                                                                                                                                   
udsoncan.setup_logging()                                                                                                                                           
conn = IsoTPSocketConnection('vcan0', isotp.Address(isotp.AddressingMode.Normal_11bits, rxid=0x123, txid=0x733))                                                   
with Client(conn, request_timeout=2, config=TJZ_config) as client:                                                                                                 
    try:                                                                                                                                                           
        # client.ecu_reset(ECUReset.ResetType.hardReset)  # HardReset = 0x01                                                                                       
        client.change_session(DiagnosticSessionControl.Session.defaultSession)  # integer with value of 3                                                          
        client.unlock_security_access(TJZ_config.debug_level)  # Fictive security level. Integer coming from fictive lib, let's say its value is 5                 
        client.write_data_by_identifier(udsoncan.DataIdentifier.VIN,'ABC123456789')  # Standard ID for VIN is 0xF190. Codec is set in the client configuration     
        print('Vehicle Identification Number successfully changed.')                                                                                               
                                                                                                                                                                   
    except NegativeResponseException as e:                                                                                                                         
        print('Server refused our request for service %s with code "%s" (0x%02x)' % (                                                                              
        e.response.service.get_name(), e.response.code_name, e.response.code))                                                                                     
    except (InvalidResponseException, UnexpectedResponseException) as e:                                                                                           
        print('Server sent an invalid payload : %s' % e.response.original_payload)

pylessard, 2024-09-14:
You copied the exmaple as is and used an RXID of 0x123 but your device respond on 0x73B

mpersisth, 2024-09-14:
It's done, and thank you very much.

closed 2024-09-14 by pylessard