pylessard/python-udsoncan #246: how to use canfd with udsoncan

mozore, 2024-09-29:
I'm trying to use zlgcan canfd with udsoncan, but i meet some error,how can i handle it?

description:
send start: 
Timestamp:        0.000000    ID:      723    S Rx     F          DL:  3    02 10 03
send end
For SingleFrames conveyed on a CAN message with data length (CAN_DL) > 8, length should be encoded on byte #1 and byte #0 should be 0x00
For SingleFrames conveyed on a CAN message with data length (CAN_DL) > 8, length should be encoded on byte #1 and byte #0 should be 0x00
For SingleFrames conveyed on a CAN message with data length (CAN_DL) > 8, length should be encoded on byte #1 and byte #0 should be 0x00
For SingleFrames conveyed on a CAN message with data length (CAN_DL) > 8, length should be encoded on byte #1 and byte #0 should be 0x00
Timestamp:     4605.683600    ID:      72b    S Rx     F          DL:  8    06 50 03 00 32 01 f4 00     Channel: 0
[TimeoutException] : Did not receive response in time. Global request timeout time has expired (timeout=5.000 sec)
Traceback (most recent call last):
  File "D:\4_workspace\project\main.py", line 40, in <module>
    main()
  File "D:\4_workspace\project\main.py", line 22, in main
    boot.test_session()
  File "D:\4_workspace\project\bootloader\bootloader.py", line 233, in test_session
    response = self.client_phy_15.change_session(3)
  File "D:\4_workspace\project\udsoncan\client.py", line 167, in decorated
    return func(self, *args, **kwargs)
  File "D:\4_workspace\project\udsoncan\client.py", line 225, in change_session
    response = self.send_request(req)
  File "D:\4_workspace\project\udsoncan\client.py", line 2216, in send_request
    raise TimeoutException('Did not receive response in time. %s time has expired (timeout=%.3f sec)' %
udsoncan.exceptions.TimeoutException: Did not receive response in time. Global request timeout time has expired (timeout=5.000 sec)

pylessard, 2024-09-29:
The problem is the device you are talking with. The implementation of the ISOTP protocol is incorrect.
As the message says, When using CAN FD with frames longer than 8 bytes, single frame length must be encoded on the second byte.

pylessard, 2024-09-29:
See this diagram taken from ISO-15765 2016
[image]
[image]

mozore, 2024-09-29:
Thanks for your answer, i got it. But i still want to know is there anyway i can use your module, can you give some advise?

pylessard, 2024-09-29:
udsoncan can be tied to any underlying layer. If you have a mean to transport data between your device and your computer, you can use udsoncan. The istop package will not accept what your device sends because it is not speaking the correct language.

If you can disable CAN FD, there are chances that your device starts being correct.
Otherwise, you can fix your device (or contact the manufacturer)
You can also fork my isotp package and add a workaround to accept the wrong messages

mozore, 2024-09-29:
Thanks again, I will close this issue.

closed 2024-09-29 by mozore