pylessard/python-udsoncan #251: 'Params' object has no attribute 'wait_func' and no __dict__ for setting new attributes

momet83, 2024-11-28:
HI @pylessard, 
thank for publishing this great work.
I'm traying to use your code but I get this outputs (see below):

**Output:;**
_Exception has occurred: AttributeError
'Params' object has no attribute 'wait_func' and no __dict__ for setting new attributes
  File "C:\Users\MohammadTarsani\Documents\4_geihub\python-udsoncan-master\python-udsoncan-master\udsoncan\beispiel3.py", line 46, in <module>
    stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)               # Network/Transport layer (IsoTP protocol)
AttributeError: 'Params' object has no attribute 'wait_func' and no __dict__ for setting new attributes_

I checked the file protocol.py and the value wait_fun does exist!

cann you please support here. thanks in Advance!

**hier ist the Exempl code**
_# Note that, in order to run this code, both ``python-can`` and ``can-isotp`` must be installed.
import udsoncan
import isotp
import can
from can.interfaces.pcan import PcanBus
from udsoncan.connections import PythonIsoTpConnection
from udsoncan.client import Client
from udsoncan.Response import Response
from udsoncan.exceptions import *
from udsoncan.services import *
from udsoncan import Dtc, DidCodec
from test.ClientServerTest import ClientServerTest
import struct

udsoncan.setup_logging()

status_mask = 0x2F

# Refer to isotp documentation for full details about parameters
isotp_params = {
    'stmin' : 32,                          # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
    'blocksize' : 8,                       # Request the sender to send 8 consecutives frames before sending a new flow control message
    'wftmax' : 0,                          # Number of wait frame allowed before triggering an error
    'll_data_length' : 8,                  # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
    'tx_padding' : 0,                      # Will pad all transmitted CAN messages with byte 0x00. None means no padding
    'rx_flowcontrol_timeout' : 1000,        # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
    'rx_consecutive_frame_timeout' : 1000, # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
    'squash_stmin_requirement' : False     # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
}

bus = PcanBus("PCAN_USBBUS1", bitrate=500000)                                           # Link Layer (CAN protocol)
tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x78A, rxid=0x7CA) # Network layer addressing scheme
stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)               # Network/Transport layer (IsoTP protocol)
conn = PythonIsoTpConnection(stack)                                                 # interface between Application and Transport layer

with Client(conn, request_timeout=10) as client:                                     # Application layer (UDS protocol)

    DTC_list = client.get_dtc_by_status_mask(status_mask)
    print("DTC List : ", DTC_list)
    print("1234:", response.service_data.dtcs)_

pylessard, 2024-11-28:
wait_func ha been added in v2.x, make sure to use that major version
Also, the error is triggered inside a file that I did not write. Have you copy pasted the original code in a file you maintain yourself? If that is so, I can't do much.

momet83, 2024-11-28:
I download your zip file and used it as it is. 
I made debuging until the line "stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)               # Network/Transport layer (IsoTP protocol)" then it comes the Error with wait_func

pylessard, 2024-11-28:
What version do you use?

momet83, 2024-11-28:
this: https://github.com/pylessard/python-udsoncan/tree/master

pylessard, 2024-11-28:
wait_func is a parameter in the can-isotp module. Not part of udsoncan.
Sorry nothing makes sense. The stack trace clearly shows that you have addeda file named "beispiel3.py"

I think you may have can-isotp v1.x installed on your machine

momet83, 2024-11-29:
I have added a file “example3.py” to send a request to the server. I didn't see an example of the application layer in your project and that's the only thing I added.

here are the packages what I inslled:
            Package           Version
            ----------------- -------
            can-isotp         2.0.6
            cffi              1.17.1
            colorama          0.4.6
            identify          2.6.3
            iniconfig         2.0.0
            packaging         24.2
            pip               24.3.1
            pluggy            1.5.0
            pycparser         2.22
            pytest            8.3.3
            python-can        4.5.0
            setup_cfg_fmt     2.7.0
            setup-py-upgrade  1.3.1
            setuptools        69.2.0
            somepackage       1.2.3
            typing_extensions 4.12.2
            udsoncan          1.23.2
            ukkonen           1.0.1
            wrapt             1.17.0

pylessard, 2024-12-03:
I can only help of you share the content of the failing file

momet83, 2024-12-03:
thanks a lot! 
after I have re-installed VS code and all Python Installation... I didn't get the Error

closed 2024-12-03 by pylessard