pylessard/python-udsoncan #252: When NRC78 is used, the server shall always send a final response

amsyrena, 2024-12-05:
[image]
[image]

should return real response even “spr_used”？

amsyrena, 2024-12-05:
but maybe when set `suppress_positive_response.enabled = True`, expect no response, so time pending is not existed.

pylessard, 2024-12-05:
I see.
That will be hard to implement. There are fundamental design choice that prevents handling this. Essentialy the synchronous design of the client. I can't reliably know if I should wait or not, unless I wait.

Would need a more complex client that has a background thread keeping track of each requests and returning Futures to the user instead of a response.

That is not in my plan sort term, nor mid term. My time is dedicated to another project right now

amsyrena, 2024-12-10:
https://github.com/pylessard/python-udsoncan/pull/254 please check, thanks

amsyrena, 2024-12-10:
We really need this change，thank you so much！