pylessard/python-udsoncan #253: Incorrect Exception Message in send_request() of client.py

6reend0g, 2024-12-05:
When reviewing the client.py code at line 2216, I noticed the following:

`raise TimeoutException('**Did** not receive response in time. %s time has expired (timeout=%.3f sec)' %)`

The term "Did" seems to be a typo or a misrepresentation. Based on the context, "Sid" might be more appropriate. However, to make the message more accurate and universally applicable, I suggest replacing it with "Service."

Here’s why:
This exception is raised whenever a diagnostic request times out, regardless of the specific Sid (e.g., 22 or 2E services, which are related to DIDs). The term "Service" would better reflect that the timeout applies broadly to any diagnostic service, not just DID-related ones.

pylessard, 2024-12-05:
Did is the verb "to do", past tense

6reend0g, 2024-12-05:
Fine, this is a little bit misleading for me . XD

pylessard, 2024-12-05:
I can update the message to use a more functional language, something like "Failed to receive a response in time". Does that seems better to you?

6reend0g, 2024-12-05:
I prefer this expression, 'Failed' makes it seem more like error handling.
By the way, there is one more question want to bother you.when using the isotp as conn of a Client, It`s hard to send messages via physical address and functional address using the same Client, whether you have a better method to solve this,or maybe there is a library better than isotp when using DoCAN.

6reend0g, 2024-12-06:
> I prefer this expression, 'Failed' makes it seem more like error handling. By the way, there is one more question want to bother you. when using the isotp as conn of a Client, it`s hard to send [...]

I figured it out: I had created two buses for each isotp stack in my previous program. This might have caused a preemption issue between the two buses. When I created only one bus to handle both isotp stacks, the problem was resolved.

In the end, thank you for your reply. Have a good day.

closed 2025-03-20 by pylessard