pylessard/python-udsoncan #274: "How to Continuously Send 3E Service Requests like CANOE: Is There a Configurable Switch or Is Multi-threading Required?"

zhikangfan01, 2025-05-22:
Assuming I am designing the BOOTLOAD function for a UI page, there is a window period after entering a certain service. If the session is not maintained at this time, the service will exit. Is there any configuration locally that can support continuously sending session 3E 80? If not, I will write my own multi-threaded execution. Does the client support multiple processes and will it trigger the capture logic error of invalid messages? Could you please guide and explain

martinjthompson, 2025-05-22:
If it were me, for something this simple, I would just setup a raw CAN message of `02 3e 80` with python-can and send it using a Broadcast Manager https://python-can.readthedocs.io/en/stable/bcm.html.

pylessard, 2025-05-22:
There is no feature for that.
You can have your own mechanism for concurrent execution. 
Be careful as the the client is not thread safe. You may want 2 clients.

You may want to read this too : https://github.com/pylessard/python-udsoncan/issues/255

closed 2025-09-09 by pylessard